xbianonpi/xbian #872: Raspberry Pi 4 + Docker run multiple cpus

josephholleman, 2019-12-07:
I have an application that runs docker to execute a spring boot app while also running ffmpeg commands for audio transcoding. Using the raspberry pi 4 (4gb ram) the container is only using 1 CPU which is causing the app to be slow. Attempting to set multiple CPUs for the container causes an error. What kernel is available which allows for multi cpu support with docker + [raspberry pi 4](https://docsbay.net/inside-raspberry-pi-4)?

My docker file is as follows;

```
FROM openjdk:11-jdk-slim-stretch
VOLUME /tmp

ADD ./target/app-1.0.0-SNAPSHOT.jar /app.jar
EXPOSE 5001 5002

ENV JAVA_OPTS=""
ENTRYPOINT [ "sh", "-c", "java $JAVA_OPTS -Djava.security.egd=file:/dev/./urandom -jar /app.jar" ]
```
I am attempting to run the container with the following command;

`docker run -p 5000:5000 --memory=200mb --cpuset-cpus="0-3" my_app`

But I have the following warnings which prevent running multi cpu images;

```
WARNING: No memory limit support
WARNING: No swap limit support
WARNING: No kernel memory limit support
WARNING: No oom kill disable support
WARNING: No cpu cfs quota support
WARNING: No cpu cfs period support
```

What OS/Kernel would allow me to run multi cpu support on the Raspberry Pi 4?

mkreisl, 2019-12-09:
Sorry, this is neither an XBian issue nor Docker support here

You'll have to build your own kernel

closed 2019-12-09 by mkreisl

mkreisl, 2019-12-09:
Btw, Docker does not report those warnings with my devel 5.4.2+ kernel:

```
root@kmxbilr2 ~ # docker info
Client:
 Debug Mode: false

Server:
 Containers: 0
  Running: 0
  Paused: 0
  Stopped: 0
 Images: 0
 Server Version: 19.03.5
 Storage Driver: btrfs
  Build Version: Btrfs v4.20.1 
  Library Version: 102
 Logging Driver: json-file
 Cgroup Driver: cgroupfs
 Plugins:
  Volume: local
  Network: bridge host ipvlan macvlan null overlay
  Log: awslogs fluentd gcplogs gelf journald json-file local logentries splunk syslog
 Swarm: inactive
 Runtimes: runc
 Default Runtime: runc
 Init Binary: docker-init
 containerd version: b34a5c8af56e510852c35414db4c1f4fa6172339
 runc version: 3e425f80a8c931f88e6d94a8c831b9d5aa481657
 init version: fec3683
 Security Options:
  seccomp
   Profile: default
 Kernel Version: 5.4.2+
 Operating System: XBian 1.0 (knockout)
 OSType: linux
 Architecture: armv7l
 CPUs: 4
 Total Memory: 3.564GiB
 Name: kmxbilr2
 ID: KRLN:3MEB:L3GN:GB6C:7ZDQ:LIB3:CMW3:XSEW:7FH2:NPGC:T337:BO5W
 Docker Root Dir: /var/lib/docker
 Debug Mode: false
 Registry: https://index.docker.io/v1/
 Labels:
 Experimental: false
 Insecure Registries:
  127.0.0.0/8
 Live Restore Enabled: false

root@kmxbilr2 ~ # 
```