xbianonpi/xbian #874: Hifiberry DAC does not work

JKocerka, 2020-03-08:
RaspberryPi 3
After a recent update a Hifiberry DAC+ card does not work any more:
dmesg: 
```sh
[   16.859067] snd-rpi-hifiberry-dacplus soc:sound: ASoC: failed to init link HiFiBerry DAC+: -517
```
aplay: 
```sh
xbian@xbian ~ $ aplay -l
aplay: device_list:272: no soundcards found...
```
The same (similar?) issue was reported recently for libreELEC:
https://forum.libreelec.tv/thread/21404-hifiberry-module-snd-soc-hifiberry-dacplus-broken-in-kernel-5-4-18/
and it was fixed few days ago:
https://forum.kodi.tv/showthread.php?tid=343068&pid=2927923#pid2927923

JKocerka, 2020-03-09:
Looks like a kernel issue: 
https://github.com/raspberrypi/linux/issues/3481

Fix is already available also for 5.4.x branch:
https://github.com/raspberrypi/linux/commit/894dece014e4df68c9f6a11a0761fba1fcfc844f

mkreisl, 2020-03-09:
Should be fixed with latest kernel build (5.4.24+)

closed 2020-06-14 by mkreisl