xbianonpi/xbian #876: Unable to "Set content" after adding video sources

morluf, 2020-04-14:
Greetings, people of xbian!

I'm having a bit of a problem with my fresh install and thought that you might be able help here (I was initially gonna open a thread on the forums instead of a bug report on your git, but apparently signing up and activating my account there wasn't enough to gain thread-posting privileges yet).

**The setup**:

Hardware: Raspberry 2B v1.1 (default clocking etc.).
Build: Kodi 18.6 Git: 20200328-0637458-dirty.
Peripherals: HDMI; LAN; BT-Keyboard USB dongle; 5V/3A power adapter.

**The issue**:

After completing the first launch wizard I updated my region settings and tried to add movies and TV shows stored on a NAS in my home network to my library. Adding the source (via SMB) works flawlessly, both from the main menu as well as the library settings (various video files I tried play when selected via File Manager, which to me suggests that this might not be a simple connection issue).
When I was about to set the correct content type I realized that xbian only allows me to choose "Movies", after which the menu changes to let me select the information provider (IMDB, TVDB etc) and save these changes as expected. When I try and select "Music videos" or "TV Shows" the behavior is the same as if I'd selected "None": no changes in appearance, nothing to select, nothing to apply. Yet after selecting either category, exiting the menu and going back in, the correct(!) selection is highlighted, even though it had no effects, and the context menu in the video source selection still says "Set content type" instead of "Change contents".

It's also not possible to change an existing source's contents; I was able to add my movies just fine, but going back and trying to change their source's settings I ran into exactly the same issue.

**Replication** 

Just to be safe I re-downloaded the image file, used a different SD card and even my Raspberry 3B+, with exactly the same results. Re-flashing the new image on the initial setup also saw no changes.

**Further steps**

Now the question is obviously: Am I missing something? Is it possible I made a wrong turn somewhere? What else could I try? Is there any additional info I could have provided? Any specific logs I maybe should have attached in the first place?

Thanks in advance, cheers!

mkreisl, 2020-04-14:
Hi, I read your post tree times and still having absolutely no idea what exactly your problem is

Sorry, but at least I can change the content as often as I want

BTW, what's your nick at the forum?

morluf, 2020-04-15:
That's what I get for trying to report a bug after pulling an all-nighter...

I apologize - I have edited my post and included visual aids. Hope it makes more sense now.

mkreisl, 2020-04-15:
Again, I do not know what you're doing wrong.

I tested it again with a fresh kodi profile (removed /home/xbian/.kodi), but I can change content to all types I want without any problem

morluf, 2020-04-15:
That's so weird. 
I even tried it a fourth time, re-downloading and re-flashing the image and taking care not to change a single default setting (apart from the keyboard layout) before trying to set my content.

Since you are not able to reproduce my issue it would appear to be either an operating error on my part or a faulty set-up. Thanks anyway for checking, and sorry for the inconvenience!

closed 2020-04-15 by morluf