xbianonpi/xbian #878: Packaging problem broke XBMC

peci1, 2020-04-28:
After latest update of `xbian-package-xbmc` on my Raspi 3, Kodi stopped working.

The problem was a dangling symlink:

```
$ ll /usr/local/lib/kodi
drwxr-xr-x 1 root root      274 Apr 28 14:59 .
drwxr-xr-x 1 root root      176 Apr  5 10:03 ..
drwxr-xr-x 1 root root     1926 Apr 28 14:56 addons
lrwxrwxrwx 1 root root        8 Apr 28 14:56 kodi.bin -> kodi-gbm
-rwxr-xr-x 1 root root 26539564 Mar 29 22:46 kodi-rbpi
drwxr-xr-x 1 root root       14 Oct  8  2019 system
```

As you can see, `kodi-gbm` disappeared. Even after `apt install --reinstall xbian-package-xbmc`.

So I downloaded the latest release and the previous one, and look:

```
$ dpkg -c xbian-package-xbmc_18.6-1585514714_armhf.deb | grep -v addons | grep lib/kodi
drwxr-xr-x root/root         0 2020-03-29 22:46 ./usr/local/lib/kodi/
drwxr-xr-x root/root         0 2020-03-29 21:51 ./usr/local/lib/kodi/system/
drwxr-xr-x root/root         0 2020-03-29 21:51 ./usr/local/lib/kodi/system/players/
drwxr-xr-x root/root         0 2020-03-29 22:46 ./usr/local/lib/kodi/system/players/VideoPlayer/
-rwxr-xr-x root/root    207600 2020-03-29 22:46 ./usr/local/lib/kodi/system/players/VideoPlayer/libdvdnav-arm.so
-rwxr-xr-x root/root  26539564 2020-03-29 22:46 ./usr/local/lib/kodi/kodi-rbpi
-rwxr-xr-x root/root  32114108 2020-03-29 22:46 ./usr/local/lib/kodi/kodi-gbm
$ dpkg -c xbian-package-xbmc_18.6-1585552801_armhf.deb | grep -v addons | grep lib/kodi
drwxr-xr-x root/root         0 2020-03-30 09:25 ./usr/local/lib/kodi/
drwxr-xr-x root/root         0 2020-03-30 06:28 ./usr/local/lib/kodi/system/
drwxr-xr-x root/root         0 2020-03-30 06:28 ./usr/local/lib/kodi/system/players/
drwxr-xr-x root/root         0 2020-03-30 09:25 ./usr/local/lib/kodi/system/players/VideoPlayer/
-rwxr-xr-x root/root    227756 2020-03-30 09:25 ./usr/local/lib/kodi/system/players/VideoPlayer/libdvdnav-arm.so
-rwxr-xr-x root/root  31585604 2020-03-30 09:25 ./usr/local/lib/kodi/kodi-rbpi
```

Apparently, the `kodi-gbm` file somehow disappeared from the distributed package.

I downloaded the package from

```
$ apt-cache policy xbian-package-xbmc
xbian-package-xbmc:
  Installed: 18.6-1585514714
  Candidate: 18.6-1585552801
  Version table:
     18.6-1585552801 500
        500 http://apt.xbian.org stable/rpi-buster armhf Packages
 *** 18.6-1585514714 500
        500 http://apt.xbian.org stable/rpi2-buster armhf Packages
        100 /var/lib/dpkg/status
```

peci1, 2020-04-28:
Or is it because the newer package comes from `rpi-buster`? Should I remove `rpi-buster` from my sources list? I think I'm using the deafult one with staging:

```
$ cat /etc/apt/sources.list.d/xbian.list 
deb http://apt.xbian.org/      stable  main rpi2-buster    rpi2-buster    armv7l-buster    rpi-buster    armv7l-buster
deb http://apt.xbian.org/  staging main rpi2-buster    rpi2-buster    armv7l-buster    rpi-buster    armv7l-buster
#deb http://apt.xbian.org/  devel   main rpi2-buster    rpi2-buster    armv7l-buster    rpi-buster    armv7l-buster
```

mkreisl, 2020-04-28:
This is your fault: you have to remove all rpi-buster entries from your xbian.list and then reinstall xbian-package-xbmc

peci1, 2020-04-28:
Okay. So maybe this would need an automated solution that would edit the xbian.list based on which RPI is the system running on?

mkreisl, 2020-04-28:
there is one  in maintenance script in package xbian-package-repo

peci1, 2020-04-28:
I was wrong, this is an RPi 4, does it change something?

mkreisl, 2020-04-28:
No

peci1, 2020-04-28:
Okay, so was this a consequence of a bad update/upgrade?

mkreisl, 2020-04-28:
I already told: I have **no** idea!
The script never has been changed for years

peci1, 2020-04-28:
Okay. Feel free to close this bug if you think it should be closed. If it starts happening to more people, they'll hopefully find it and it can be reopened.

closed 2020-06-14 by mkreisl