xbianonpi/xbian #880: Тo picture on TV (HDMI) when booting from USB on rpi3b+

DejavuDmitry, 2020-07-10:
Some stange behavior...
After writing image to USB SSD (KingDian S280-120) and booting I can't see anything on my TV.
HDMI cable didn't unplugged, only booting media was switched.
After restoring microSD card image appears again. WTF?

But I can connect to rpi3b+ via ssh and with USB SSD I have this output:
```bash
xbian@xbian ~ $ tvservice -s
state 0x40000 [NTSC 4:3], 720x480 @ 60.00Hz, interlaced
xbian@xbian ~ $ tvservice -l
1 attached device(s), display ID's are : 
Display Number 3, type Composite
```
And via microSD (with having picture):
```bash
xbian@xbian ~ $ tvservice -s
state 0xa [HDMI CEA (16) RGB lim 16:9], 1920x1080 @ 60.00Hz, progressive
xbian@xbian ~ $ tvservice -n
device_name=PHL-PHILIPS_FTV
xbian@xbian ~ $ vcgencmd display_power
display_power=-1
xbian@xbian ~ $ tvservice -l
1 attached device(s), display ID's are : 
Display Number 2, type HDMI 0
```

``` bash
inux xbian 5.4.50+ #1 SMP PREEMPT Sat Jul 4 13:24:54 CEST 2020 armv7l
--------------------------------------------------------------------
       Y88b d88P  888888b.   8888888        d8888 888b    888
        Y88o88P   888  .88P    888        d88P888 88888b  888
         Y888P    8888888K.    888       d88P 888 888Y88b 888
         d888b    888  "Y88b   888      d88P  888 888 Y88b888
       d88P Y88b  888   d88P   888    d8888888888 888   Y8888
      d88P   Y88b 8888888P"  8888888 d88P     888 888    Y888
--------------------------------------------------------------------
           XBian 20200508-0 - Bleeding Edge, 2012 - 2020
```

Any suggestions?

mkreisl, 2020-07-11:
I'm sorry, I can't help you there. These strange problems occur sporadically, I can't say what's causing it.
With one user even the HDMI output was defective

DejavuDmitry, 2020-07-18:
I just "reburn" image and all works.
But after first upgrade and rebooting I have again "headless machine".
Here is /var/log/apt/history.log
`Start-Date: 2020-07-18  12:01:31
Commandline: apt-get install -y -o Dpkg::Options::=--force-confdef -o Dpkg::Options::=--force-confold linux-image-bcm2836=5.4.50+-1593862545 xbian-package-firmware=9-1594745601 xbian-package-vnc-server=2.5.4-1592583499 xbian-package-xbmc=18.7-1592878545
Requested-By: xbian (1000)
Upgrade: linux-image-bcm2836:armhf (5.4.44+-1591466742, 5.4.50+-1593862545), xbian-package-vnc-server:armhf (2.5.4-1590509896, 2.5.4-1592583499), xbian-package-firmware:armhf (9-1575993042, 9-1594745601), xbian-package-xbmc:armhf (18.7-1590242823, 18.7-1592878545)
End-Date: 2020-07-18  12:03:22`

Inxi output:
```bash
xbian@xbian ~ $ inxi -F                                                                         System:    Host: xbian Kernel: 5.4.50+ armv7l bits: 32 Console: tty 1                                      Distro: XBian 1.0 (knockout)
Machine:   Type: ARM Device System: Raspberry Pi 3 Model B Plus Rev 1.3 details: BCM2835
           rev: a020d3 serial: 000000003cc522d3
Argument "Raspberry Pi 3 Model B Plus Rev 1.3" isn't numeric in sprintf at /usr/bin/inxi line 6969.
CPU:       Topology: Quad Core model: ARMv7 v7l variant: cortex-a53 bits: 32 type: MCP
           Speed: 700 MHz min/max: 700/1400 MHz Core speeds (MHz): 1: 700 2: 700 3: 700
           4: 700
Graphics:  Device-1: bcm2708-fb driver: bcm2708_fb v: kernel
           Device-2: bcm2835-hdmi driver: N/A
           Display: server: No display server data found. Headless machine? tty: 96x35
           Message: Unable to show advanced data. Required tool glxinfo missing.
Audio:     Device-1: bcm2835-audio driver: bcm2835_audio
           Device-2: bcm2835-hdmi driver: N/A
           Sound Server: ALSA v: k5.4.50+
Network:   Device-1: Standard Microsystems type: USB driver: lan78xx
           IF: eth0 state: up speed: 100 Mbps duplex: full mac: b8:27:eb:c5:22:d3
           IF-ID-1: wlan0 state: down mac: b8:27:eb:90:77:86
Drives:    Local Storage: total: 111.79 GiB used: 2.85 GiB (2.6%)
           ID-1: /dev/sda type: USB vendor: KingDian model: P10 120GB size: 111.79 GiB
Partition: ID-1: / size: 111.45 GiB used: 2.81 GiB (2.5%) fs: btrfs dev: /dev/sda2
           ID-2: /boot size: 99.8 MiB used: 42.6 MiB (42.7%) fs: vfat dev: /dev/sda1
           ID-3: /home size: 111.45 GiB used: 2.81 GiB (2.5%) fs: btrfs dev: /dev/sda2                     ID-4: swap-1 size: 244.5 MiB used: 0 KiB (0.0%) fs: swap dev: /dev/sda3              Sensors:   Missing: Required tool sensors not installed. Check --recommends
Info:      Processes: 141 Uptime: 9m Memory: 933.7 MiB used: 436.0 MiB (46.7%)
           gpu: 256.0 MiB Init: N/A runlevel: 2 Shell: bash inxi: 3.0.32
```

mkreisl, 2020-07-18:
We're not getting anywhere like this. I'm sorry.
Instead of presenting such a history log, you should rather install package by package and check if the system still works afterwards

And like I said, I can't help you there, it seems that something is wrong with your hardware configuration