xbianonpi/xbian #881: Error while installing xbian in raspberry pi 4 8b RAM

venkatmakam, 2020-07-14:
Hello,
I tried to install xbian in my raspberry pi 4.while booting it throws error
`This board required newest software`.
Also i used noobs xbian image.but xbian os is not listed in noobs os list.
Any help?

mkreisl, 2020-07-14:
Which image did you used? I never seen this message
And, you have to use PINN

mkreisl, 2020-07-14:
https://community.volumio.org/t/raspberry-pi-4-8gb-cannot-install-this-board-requires-newer-software/39536
If this is is the message you get, firmware files are too old.

I have to update this package (those files are located in xbian-package-firmware) and new image has to be build at Friday

venkatmakam, 2020-07-14:
> https://community.volumio.org/t/raspberry-pi-4-8gb-cannot-install-this-board-requires-newer-software/39536
> If this is is the message you get, firmware files are too old.
> 
> I have to update this package (those files are located in xbian-package-firmware) and new image has to be build at Friday

Yes exactly, same error i am facing.

mkreisl, 2020-07-14:
Ok, if you have an older Pi (2,,3,4(<8GBRAM)), you could put this sd card into this one and update the package xbian-package-firmware (available about in one hour).
After updating this package everything should be fine

venkatmakam, 2020-07-14:
> Ok, if you have an older Pi (2,,3,4(<8GBRAM)), you could put this sd card into this one and update the package xbian-package-firmware (available about in one hour).
> After updating this package everything should be fine

I have RPI2.I will try tomorrow.Thank you.
Also i used the PINN installer from
https://sourceforge.net/projects/xbian/files/release/XBian_by_PINN_3.3.4_rpi.img.gz/download,but i did not see xbian either in general or any other tabs. Am i missing something?

mkreisl, 2020-07-14:
Please use PINN from our download page

venkatmakam, 2020-07-14:
Yes i saw rpi4 missing.I thought of adding manually,but could not tried.:)
So if i want to install xbian along with PINN i need to wait till image is updated.
But mean while i can use the [solution](https://github.com/xbianonpi/xbian/issues/881#issuecomment-658309859),for installing xbian alone.Am i right?

mkreisl, 2020-07-14:
Yes, that's the standard way.

And, having just updated XBian's PINN version to 3.3.4.2, should work now again with Pi4 again

venkatmakam, 2020-07-15:
Hi i did xbian-package-firmware in RP2 .But i am facing same issue in RPI4.
[image]
[image]

venkatmakam, 2020-07-15:
Yes it was updated correctly after changing the time zone.Now i am able to boot in RPI4.Thanks.

mkreisl, 2020-07-15:
Thanks for confirming

closed 2020-07-15 by mkreisl