xbianonpi/xbian #885: [rpi4] [Kodi] XBian settings applet crashes, missing python dependency

eremem, 2020-09-21:
A fresh installation on rpi4 from XBian_2020.09.18_rpi4.img.
Opening the XBian settings applet shows the usual loading progress dialog and nothing more.
In kodi.log I could find the following:

```
2020-09-21 20:41:59.823 T:2422014160   ERROR: Traceback (most recent call last):
2020-09-21 20:41:59.823 T:2422014160   ERROR:   File "/usr/local/share/kodi/addons/plugin.xbianconfig/default.py", line 258, in <module>
2020-09-21 20:41:59.824 T:2422014160   ERROR:     win.show()
2020-09-21 20:41:59.824 T:2422014160   ERROR:   File "/usr/local/share/kodi/addons/plugin.xbianconfig/default.py", line 114, in show
2020-09-21 20:41:59.824 T:2422014160   ERROR:     self.onShow()
2020-09-21 20:41:59.824 T:2422014160   ERROR:   File "/usr/local/share/kodi/addons/plugin.xbianconfig/default.py", line 149, in onShow
2020-09-21 20:41:59.824 T:2422014160   ERROR:     catmodule = __import__('%s.%s' % (CATEGORY_PATH, module), globals(), locals(), [module])
2020-09-21 20:41:59.824 T:2422014160   ERROR:   File "/usr/local/share/kodi/addons/plugin.xbianconfig/categories/59_extra.py", line 8, in <module>
2020-09-21 20:41:59.825 T:2422014160   ERROR:     class extra(Category):
2020-09-21 20:41:59.825 T:2422014160   ERROR:   File "/usr/local/share/kodi/addons/plugin.xbianconfig/categories/59_extra.py", line 9, in extra
2020-09-21 20:41:59.825 T:2422014160   ERROR:     SETTINGS = extrasetting.extra()
2020-09-21 20:41:59.825 T:2422014160   ERROR:   File "/usr/local/share/kodi/addons/plugin.xbianconfig/categories/extra/__init__.py", line 16, in extra
2020-09-21 20:41:59.825 T:2422014160   ERROR:     setarray = __import__(module, globals(), locals(), [module])
2020-09-21 20:41:59.825 T:2422014160   ERROR:   File "/usr/local/share/kodi/addons/plugin.xbianconfig/categories/extra/01_tvheadend.py", line 2, in <module>
2020-09-21 20:41:59.828 T:2422014160   ERROR:     from builtins import map
2020-09-21 20:41:59.829 T:2422014160   ERROR: ImportError: No module named builtins
2020-09-21 20:41:59.829 T:2422014160   ERROR: Exception in xbianSettingWindow None
2020-09-21 20:41:59.911 T:2422014160 WARNING: CPythonInvoker(7, /usr/local/share/kodi/addons/plugin.xbianconfig/default.py): the python script "/usr/local/share/kodi/addons/plugin.xbianconfig/default.py" has left several classes in memory that we couldn't clean up. The classes include: N9XBMCAddon9xbmcaddon5AddonE,N9XBMCAddon7xbmcgui6DialogE
```
A dependency seems to be missing:
`2020-09-21 20:41:59.829 T:2422014160   ERROR: ImportError: No module named builtins`

Python is not my field of expertise, but a quick google search provides a possible solution:
`sudo apt -y install python-future`

It fixed the problem for me, but I guess the image shouldn't be missing any required packages.

mkreisl, 2020-09-21:
Seems you have **xbiian-package-tvheadend** installed

eremem, 2020-09-21:
Yes, that's right.

mkreisl, 2020-09-21:
So, seems have to fix this fix: https://github.com/xbianonpi/xbian-package-tvheadend/commit/8a036e4fe2233cfbf7e287f024b5bb05555dc3f4

mkreisl, 2020-09-21:
Maybe this would help to fix it, could you please try this and deinstall python-future again?

`File /usr/local/share/kodi/addons/plugin.xbianconfig/categories/extra/01_vncserver.py`

```
from __future__ import print_function

try:
    from builtins import map
    import itertools.ifilter as filter
except ImportError:
    pass

eremem, 2020-09-21:
I replaced the previous lines with those above but it doesn't seem to solve the problem:

```
2020-09-21 21:26:10.609 T:2802266320   ERROR: Traceback (most recent call last):
2020-09-21 21:26:10.609 T:2802266320   ERROR:   File "/usr/local/share/kodi/addons/plugin.xbianconfig/default.py", line 258, in <module>
2020-09-21 21:26:10.610 T:2802266320   ERROR:     win.show()
2020-09-21 21:26:10.610 T:2802266320   ERROR:   File "/usr/local/share/kodi/addons/plugin.xbianconfig/default.py", line 114, in show
2020-09-21 21:26:10.610 T:2802266320   ERROR:     self.onShow()
2020-09-21 21:26:10.611 T:2802266320   ERROR:   File "/usr/local/share/kodi/addons/plugin.xbianconfig/default.py", line 149, in onShow
2020-09-21 21:26:10.611 T:2802266320   ERROR:     catmodule = __import__('%s.%s' % (CATEGORY_PATH, module), globals(), locals(), [module])
2020-09-21 21:26:10.611 T:2802266320   ERROR:   File "/usr/local/share/kodi/addons/plugin.xbianconfig/categories/59_extra.py", line 8, in <module>
2020-09-21 21:26:10.611 T:2802266320   ERROR:     class extra(Category):
2020-09-21 21:26:10.611 T:2802266320   ERROR:   File "/usr/local/share/kodi/addons/plugin.xbianconfig/categories/59_extra.py", line 9, in extra
2020-09-21 21:26:10.612 T:2802266320   ERROR:     SETTINGS = extrasetting.extra()
2020-09-21 21:26:10.612 T:2802266320   ERROR:   File "/usr/local/share/kodi/addons/plugin.xbianconfig/categories/extra/__init__.py", line 16, in extra
2020-09-21 21:26:10.612 T:2802266320   ERROR:     setarray = __import__(module, globals(), locals(), [module])
2020-09-21 21:26:10.612 T:2802266320   ERROR:   File "/usr/local/share/kodi/addons/plugin.xbianconfig/categories/extra/01_tvheadend.py", line 2, in <module>
2020-09-21 21:26:10.612 T:2802266320   ERROR:     from builtins import map
2020-09-21 21:26:10.613 T:2802266320   ERROR: ImportError: No module named builtins
2020-09-21 21:26:10.613 T:2802266320   ERROR: Exception in xbianSettingWindow None
2020-09-21 21:26:10.700 T:2802266320 WARNING: CPythonInvoker(10, /usr/local/share/kodi/addons/plugin.xbianconfig/default.py): the python script "/usr/local/share/kodi/addons/plugin.xbianconfig/default.py" has left several classes in memory that we couldn't clean up. The classes include: N9XBMCAddon7xbmcgui6DialogE,N9XBMCAddon9xbmcaddon5AddonE
```

mkreisl, 2020-09-21:
Kodi restarted?

mkreisl, 2020-09-21:
Same issue was here, but fix worked there http://forum.xbian.org/thread-4093-post-36667.html

eremem, 2020-09-21:
Yes, I restarted Kodi.
I guess I nailed it.
In my case the problem was located in 01_tvheadend.py instead of 01_vncserver.py (I don't use the VNC server at all), as the error message suggested:
ERROR:     File "**/usr/local/share/kodi/addons/plugin.xbianconfig/categories/extra/01_tvheadend.py**", line 2, in <module>
ERROR:     from builtins import map

I moved `from builtins import map` from the 2nd line inside the `try:` clause (as in your code for 01_vncserver.py) and the applet starts w/o logging any errors.

mkreisl, 2020-09-21:
Ok, its working. Sorry, I had confused the two files.

Will fix this in **xbian-package-tvheadend** and **xbian-packacke-transmission** tomorrow

eremem, 2020-09-21:
No problem! Thank you for your assistance and the fix. :)

closed 2022-07-09 by eremem