xbianonpi/xbian #886: [rpi4] System halts on reboot

eremem, 2020-09-21:
Every time when I try to reboot the system (either from Kodi or in the command line), the system shows the following steps on the splash screen:
- rebooting...
- all killed. unmounting fs.
- **system halted.**

and shuts down:|

Installed from XBian_2020.09.18_rpi4.img.

mkreisl, 2020-09-23:
Sorry, I can't reproduce it

Tested it with fresh XBian_2020.09.18_rpi4.img installed on an external USB disk and my longterm stable XBian installed on sd-card which is always up2date, and reboot **always** works

And of course my development installations for Kodi 19 reboot all as they should

eremem, 2020-09-23:
Thank you for your check.

That's strange though, since except for some extra packages installed from the default repo(s), I have nothing but the vanilla distro running here.

I will do a fresh install in the following days and let you know if the problem occurs there or not.

eremem, 2020-09-24:
It seems to be a hardware issue - many active USB hubs back-power the RPi 4 on reboot, which prevents it from booting up. Unplugging the USB cable "magically" restores rebooting, which of course is no solution at all. Reportedly there are hardly any active USB hubs out there, which don't affect the reboot on RPi 4 (!!!). More about the issue [here](https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=244900). 

New RPi generation, new quirks. "Nice"...

mkreisl, 2020-09-24:
Yes, it is unfortunately the case that the Raspberry Pi Foundation is producing more and more messes. It started really badly with the Pi3B, and the Pi4 is not much better.
Also concerning the software development (especially around Kodi), a single tragedy

eremem, 2020-09-26:
Well, I guess that's the price of popularity. Maybe the Pi4's design is right and the rest of the world is USB-standard-incompatible. The only problem is, this way the Pi folks make a bunch of their clients disappointed if not angry. They can fight the world but can they win?...
I made a last test and disabled the Vcc pin in the hub's plug. Now the reboot _sometimes_ works - it's too unreliable anyway. Not a big problem for my current project, but a pain in the a*, should the Pi run unattended and need to be rebooted remotely...

I close this issue now.
Thank you for your support.

closed 2020-09-26 by eremem