xbianonpi/xbian #887: Having same issue as #876

seanwalter, 2020-10-12:
I am running xbian 1.0 (knockout) with Kodi 18.8 (18.8.0) Git:20200808-7631db6848-dirty, on a new clean install on a Pi 2B v2.1.

I can specify video source content type as Movies or None, but not as TV Shows or Music Videos. If I choose either of these two, the content type does not update.

In other words:

- The content type is None by default, and if I change it to TV Shows or Music Videos, it stays as None when I click OK
- If I change it to Movies, that change is preserved when I click OK.
- If I have set it to Movies and then try to change it to TV Shows or Music Videos, it stays as Movies when I click OK

This appears to be the same as issue #876 which is marked closed, but without a resolution.

Note that prior to posting this, I cleaned my database and tried again. I subsequently deleted all sources, cleaned the database and tried yet again. In all cases, the problem persisted.

Debug logfile uploaded to https://paste.kodi.tv/kusekiseka 

Pretty sure the relevant parts start around time stamp 22:01:38.380 where a few lines in the log show:

`Skin file /usr/local/share/kodi/addons/skin.estuary/xml/DialogSettings.xml loaded in 15.26ms
Alloc resources: 16.21ms  (15.58 ms skin load)
CSettingsManager: requested setting (contenttype) was not found.`

Happy to help try things out to resolve this. Thanks!

mkreisl, 2020-10-12:
I'm sorry, but we're not getting anywhere like this.

I have tested it again and everything works as it should. I can make every possible adjustment, no problem visible

seanwalter, 2020-10-12:
Thank you for checking. It feels pretty random to me, and I’m sure it’s frustrating to hear this issue coming back when you don’t see it yourself.

Are you using the Estuary skin? I am wondering if it’s a quirk between Xbian and certain skins, but this feels like a long shot. I did test with Aeon Nox Silvo, but only that and Estuary.

Also a long shot, but my media sources are all read-only SMB shares from a NAS on my LAN. Not sure if that would affect anything, but it’s probably different from many people’s setups.

I will try deleting various files over the next day or so to see if it’s just a corrupt file that can get recreated cleanly. I will also test whether these content types can be assigned correctly to local and SMB directories if the xbian user has read/write permissions.

If you are using a different skin, let me know, and of course if you have ideas for other things I can try, I’m open to suggestions.

mkreisl, 2020-10-12:
I'm using standard Esturary skin. And I was testing SMB and SFTP shares. Maybe you have a conflict with the configured 
scraper. And I don't think that r/o shares makes a difference

seanwalter, 2020-10-12:
Cool. I will change scrapers and see what happens. Right now it’s whatever xbian defaults to on installation, which I think is Universal Media Scraper. Anyway, I’ll try changing that first.

seanwalter, 2020-10-13:
Ok...problem solved! The TV Show scraper installed by default in Xbian is _The Movie Database TMDB TV show scraper_. I added The TVDB scraper from the default repo and now my TV Shows folder can be set for TV Show content type.

Similarly, I added the one Music Video scraper in the default repos (TheAudioDb.com) and after doing so I can assign a source to be content type Music Videos.

Not sure if this is strictly a Kodi thing or if it's an Xbian thing, so I will leave this issue open and @mkreisl if it should be closed feel free to do so.

mkreisl, 2020-10-13:
It is AFAIK not XBian issue

closed 2020-10-13 by mkreisl

alexsalex, 2021-10-07:
Have same issue