xbianonpi/xbian #889: Python{2,3}: RuntimeError: This module can only be run on a Raspberry Pi!

alexsalex, 2020-12-06:
Hi!

Im using the retroflag case for my mediasenter. DO not use RetroPie, just like the case and safe shutdown function.

When I run safe shutdown script its shows me this error:

```
root@mc /opt/RetroFlag # python ./SafeShutdown.py
Traceback (most recent call last):
  File "./SafeShutdown.py", line 1, in <module>
    import RPi.GPIO as GPIO
  File "/usr/lib/python2.7/dist-packages/RPi/GPIO/__init__.py", line 23, in <module>
    from RPi._GPIO import *
RuntimeError: This module can only be run on a Raspberry Pi!
root@mc /opt/RetroFlag #
```

All modules installed, latest version from repo. How can I fix this issue?

In this case I can't use GPIO in the Python on XBian... that is not good....

mkreisl, 2020-12-07:
From where does the Python stuff came from?

alexsalex, 2020-12-07:
No modifications, no updates on the system. The same issue on Python 2 and Python 3.

mkreisl, 2020-12-08:
Again, from where does the Python stuff came from ?!?

```
/opt/RetroFlag # python ./SafeShutdown.py
```

Thats neither XBian nor Debian source

mkreisl, 2020-12-08:
I once unpacked my crystal ball

You are probably using a Raspberry Pi4 and have fetched the python-rpi.gpio library from Debian repository. Unfortunately this is the old 0.6.5 version which does not know the Pi4. 
I got the current 0.7 packages and added them to our repository, which should be available everywhere in 2 hours.

So a `sudo apt-get update && sudo apt-get upgrade` should help

alexsalex, 2020-12-08:
> Again, from where does the Python stuff came from ?!?
> 
> ```
> /opt/RetroFlag # python ./SafeShutdown.py
> ...
> /usr/lib/python2.7/dist-packages/RPi/GPIO/__init__.py
> ```
> 
> That's definitely no XBian software

https://github.com/RetroFlag/retroflag-picase

Update doesn't help.

mkreisl, 2020-12-08:
If you really have the 0.7 packages installed, you have to search for the bug yourself.

This is no XBian problem and I don't have such a case, so I can't reproduce it

alexsalex, 2020-12-08:
> If you really have the 0.7 packages installed, you have to search for the bug yourself.
> 
> This is no XBian problem and I don't have such a case, so I can't reproduce it

If you install on Raspberry PI 4 the package python-rpi.gpio, then run python:

```
xbian@mc ~ $ python
Python 2.7.16 (default, Oct 10 2019, 22:02:15)
[GCC 8.3.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import RPi.GPIO as GPIO
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python2.7/dist-packages/RPi/GPIO/__init__.py", line 23, in <module>
    from RPi._GPIO import *
RuntimeError: This module can only be run on a Raspberry Pi!
>>>
```

Do you see this error?

mkreisl, 2020-12-08:
Sure, I see it.
 
I did the same on my Pi4, latest kernel and I do **not** get this error message

```
root@kmxbilr2 ~ # dpkg -l | grep GPI
ii  python3-rpi.gpio                     0.7.0-0.1~bpo10+1                   armhf        Module to control Raspberry Pi GPIO channels (Python 3)
ii  rpi.gpio-common:armhf                0.7.0-0.1~bpo10+1                   armhf        Module to control Raspberry Pi GPIO channels (common files)
root@kmxbilr2 ~ # python3
Python 3.7.3 (default, Jul 25 2020, 13:03:44) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import RPi.GPIO
>>> 
root@kmxbilr2 ~ # uname -a
Linux kmxbilr2 5.4.75+ #1 SMP PREEMPT Wed Nov 11 14:50:30 CET 2020 armv7l GNU/Linux
root@kmxbilr2 ~ # 
```

alexsalex, 2020-12-08:
> Sure, I see it.
>
> I did the same on my Pi4, latest kernel and I do **not** get this error message
>
> ```
> root@kmxbilr2 ~ # dpkg -l | grep GPI
> ii  python3-rpi.gpio [...]

Did you try it on XBian?

root@car-mc ~ # python3
Python 3.7.3 (default, Jul 25 2020, 13:03:44)
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import RPi.GPIO as GPIO
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3/dist-packages/RPi/GPIO/__init__.py", line 23, in <module>
    from RPi._GPIO import *
RuntimeError: This module can only be run on a Raspberry Pi!
>>> quit()
root@car-mc ~ # uname -a
Linux car-mc 5.4.75+ #1 SMP PREEMPT Wed Nov 11 14:55:50 CET 2020 armv7l GNU/Linux
root@car-mc ~ #

mkreisl, 2020-12-08:
I'm using XBian, nothing else

mkreisl, 2020-12-08:
Did you read this?

https://sourceforge.net/p/raspberry-gpio-python/tickets/190/

alexsalex, 2020-12-08:
> Did you read this?
> 
> https://sourceforge.net/p/raspberry-gpio-python/tickets/190/
> 
> Or if you have Pi4/8GB device read here:
> 
> https://sourceforge.net/p/raspberry-gpio-python/code/merge-requests/5/
> 
> This device seems still not be supported in official files

I have RPI4 2Gb

```
root@mc ~ # tail -n 4 /proc/cpuinfo | grep -v ^Serial
Hardware	: BCM2711
Revision	: b03112
Model		: Raspberry Pi 4 Model B Rev 1.2
root@mc ~ #
```

The issue is if I'm install the Rasbian (Raspberry OS) or Armbian everything is works without any issue.

alexsalex, 2020-12-08:
Absolutely the same system with Raspbian:

```
pi@RasPi:~ $ tail -n 4 /proc/cpuinfo | grep -v ^Serial
Hardware	: BCM2711
Revision	: b03112
Model		: Raspberry Pi 4 Model B Rev 1.2
pi@RasPi:~ $ python
Python 2.7.16 (default, Oct 10 2019, 22:02:15)
[GCC 8.3.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import RPi.GPIO as GPIO
>>> quit()
pi@RasPi:~ $ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 10 (buster)
Release:	10
Codename:	buster
pi@RasPi:~ $
```

mkreisl, 2020-12-10:
I'm sure you haven't installed the 0.7 versions yet. unfortunately, i made a stupid mistake when uploading these packages, so they weren't saved correctly.
Please try again in 3h, until then the correct packages are also safely distributed on the mirrors. And make sure that you really have installed the 0.7 version
```
root@kmxbilr2 /var/cache/apt # dpkg -l | grep gpio
ii  python3-rpi.gpio                  0.7.0-0.1~bpo10+1                   armhf        Module to control Raspberry Pi GPIO channels (Python 3)
ii  rpi.gpio-common:armhf             0.7.0-0.1~bpo10+1                   armhf        Module to control Raspberry Pi GPIO channels (common files)
root@kmxbilr2 /var/cache/apt # apt-cache policy python3-rpi.gpio
python3-rpi.gpio:
  Installed: 0.7.0-0.1~bpo10+1
  Candidate: 0.7.0-0.1~bpo10+1
  Version table:
 *** 0.7.0-0.1~bpo10+1 500
        500 http://kmcubie stable/main armhf Packages
        100 /var/lib/dpkg/status
     0.6.5-1 500
        500 http://deb.debian.org/debian buster/main armhf Packages
root@kmxbilr2 /var/cache/apt # 
```

kjeldflarup, 2021-11-03:
I see this error on my RPi4

I had hoped that these changes had been pushed to Raspbian, but it seems not. Where can I get these patches?

```
pi@pi013:/opt/labgrid-admin/gpio $ python gpio --server=9999
Traceback (most recent call last):
  File "gpio", line 4, in <module>
    from flask import Flask
ImportError: No module named flask
pi@pi013:/opt/labgrid-admin/gpio $ tail -n 4 /proc/cpuinfo | grep -v ^Serial
Hardware	: BCM2711
Revision	: c03114
Model		: Raspberry Pi 4 Model B Rev 1.4
pi@pi013:/opt/labgrid-admin/gpio $ dpkg -l | grep gpio
ii  python-rpi.gpio                0.7.0-0.1~bpo10+4                   armhf        Module to control Raspberry Pi GPIO channels (Python 2)
ii  python3-rpi.gpio               0.7.0-0.1~bpo10+4                   armhf        Module to control Raspberry Pi GPIO channels (Python 3)
ii  raspi-gpio                     0.20191001                          armhf        Dump the state of the BCM270x GPIOs
ii  rpi.gpio-common:armhf          0.7.0-0.1~bpo10+4                   armhf        Module to control Raspberry Pi GPIO channels (common files)
pi@pi013:/opt/labgrid-admin/gpio $ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 10 (buster)
Release:	10
Codename:	buster

```

mkreisl, 2021-11-03:
@kjeldflarup you are completely wrong here. Please contact a Pi-OS aka Raspbian forum with your problem (whatever that problem is). Thanks

closed 2021-11-03 by mkreisl