xbianonpi/xbian #890: GZip Raspbery pi 4 image is corrupted

flori4n, 2021-01-25:
Hello, 

I've tried the last 3 images for the raspberry pi. 
If I try to extract it directly with etcher (I seleced the .gz archive) or with the gnome archive manager I have a message who said "unexpected end of file" at around 41%. 
It seems that the "extract here" in ubuntu doesn't see/show the error and create a corrupted iso.

I will try to go back in time to try to find one who isn't corrupted.
Is someone able to extract the archive one of the 2 way I have put with this image: https://sourceforge.net/projects/xbian/files/release/XBian_Latest_rpi4.img.gz/download

Thank you

mkreisl, 2021-01-25:
Please use an older image, latest build is broken and has already been deleted

flori4n, 2021-01-25:
It  worked perfectly with this image: https://netcologne.dl.sourceforge.net/project/xbian/release/XBian_2021.01.08_rpi4.img.gz
Thank you :)

closed 2021-01-25 by flori4n