xbianonpi/xbian #893: Kodi crashes after update/upgrade

LaszloLango, 2021-03-25:
Hi,

I need some help. My install is a bit old now, so I'd like to update it, but Kodi does not start after the update and most of the services are dead too (even ssh, so I cannot restore it remotely). I am able to restore the last stable snapshot with `brtfs-auto-snapshot`, so fortunately nothing lost.
I am running xbian on a raspberry Pi3. I checked my sources.list file. I commented out the debian repository:
```
deb http://deb.debian.org/debian stretch main non-free
```
so this is what I have now:
```
deb http://apt.xbian.org/      stable  main rpi2-stretch    rpi2-stretch    armv7l-stretch
### deb http://apt.xbian.org/  staging main rpi2-stretch    rpi2-stretch    armv7l-stretch
### deb http://apt.xbian.org/  devel   main rpi2-stretch    rpi2-stretch    armv7l-stretch
```

It wants to update fewer packages after this change, but the outcome is still the same. I do not see any useful thing in the upstart logs after the update. It says a `Segmentation fault` in the xbmc.log, but no stacktrace or any other useful message. I do not get any error during the apt update. What logs should I check for further details?

More information about my setup:

```bash
$> uname -a
Linux xbian 5.4.100+ #1 SMP PREEMPT Thu Mar 4 00:52:56 CET 2021 armv7l GNU/Linux
```

```bash
$> sudo apt list --upgradable
Listing... Done
initramfs-tools/stable 0.133+xbian2 all [upgradable from: 0.130-xbian1]
initramfs-tools-core/stable 0.133+xbian2 all [upgradable from: 0.130-xbian1]
libx264-148/stable 2:0.148.2748+git97eaef2-1 armhf [upgradable from: 2:0.148.2748+git97eaef2-1]
xbian-package-config-xbmc/stable 20210319-0 armhf [upgradable from: 20190622-0]
xbian-package-firmware/stable 9.5-1610486159 armhf [upgradable from: 9-1575993042]
xbian-package-initramfs-tools/stable 20200507-0 armhf [upgradable from: 20190822-0]
xbian-package-transmission/stable 20200921-0 armhf [upgradable from: 20180331-0]
xbian-package-upstart-xbmc-bridge/stable 2.0.5 armhf [upgradable from: 2.0.2]
xbian-package-xbmc-scripts/stable 1.9.7 armhf [upgradable from: 1.9.4]
xbian-update/stable 20210101-0 armhf [upgradable from: 20200109-0]
````

```bash
dpkg -l | grep -iE "xbian|xbmc|kodi|initramfs"
ii  btrfs-progs                       4.9.1-xbian                       armhf        Checksumming Copy on Write Filesystem utilities
ii  cec-utils                         4.0.4.1~xbian-1545935332          armhf        libCEC communication Library (utility programs)
ii  initramfs-tools                   0.130-xbian1                      all          generic modular initramfs generator (automation)
ii  initramfs-tools-core              0.130-xbian1                      all          generic modular initramfs generator (core tools)
ii  kodi                              2:17.1+dfsg1-3                    armhf        Open Source Home Theatre (executable binaries)
ii  kodi-bin:armhf                    2:17.1+dfsg1-3                    armhf        Open Source Home Theatre (architecture-dependent files)
ii  kodi-data                         2:17.1+dfsg1-3                    all          Open Source Home Theatre (arch-independent data package)
ii  kodi-pvr-hts                      3.4.13-1                          armhf        Kodi PVR Addon TvHeadend Hts
ii  libavcodec58:armhf                7:4.0.3-xbian3                    armhf        FFmpeg library with de/encoders for audio/video codecs - runtime files
ii  libavutil56:armhf                 7:4.0.3-xbian3                    armhf        FFmpeg library with functions for simplifying programming - runtime files
ii  libcec                            4.0.4.1~xbian-1545935332          armhf        Meta package libCEC.
ii  libcec-dev                        4.0.4.1~xbian-1545935332          armhf        libCEC communication Library (development files)
ii  libcec4:armhf                     4.0.4.1~xbian-1545935332          armhf        libCEC communication Library (shared library)
ii  libklibc                          2.0.4-9                           armhf        minimal libc subset for use with initramfs
ii  libswresample3:armhf              7:4.0.3-xbian3                    armhf        FFmpeg library for audio resampling, rematrixing etc. - runtime files
ii  linux-image-bcm2836               4.19.90+-1576699354               armhf        Latest XBian kernel (rpi2/4.19.y 4.19.90+)
ii  mountall                          2.54+xbian1                       armhf        filesystem mounting tool
ii  python-libcec                     4.0.4.1~xbian-1545935332          armhf        Python bindings for libCEC
ii  python-pycryptodome               3.6.4-1~xbian                     armhf        cryptographic Python library (Python 2)
ii  xbian-package-config-shell        20200404-0                        armhf        Configuration utility for XBian 20200404-0
ii  xbian-package-config-xbmc         20190622-0                        armhf        Configuration utility for XBian 20190622-0
ii  xbian-package-firmware            9-1575993042                      armhf        RPI firmware (RPI2 compatible)
ii  xbian-package-initramfs-tools     20190822-0                        armhf        Initramfs handling tools 20190822-0
ii  xbian-package-lirc                20170627                          armhf        Remote controller deamon.
ii  xbian-package-repo                20191030-0                        armhf        XBian repository package 20191030-0. It installs XBian public apt keys and sources files.
ii  xbian-package-samba               20170127-0                        armhf        Windows <--> linux filesharing 20170127-0
ii  xbian-package-shairplay           1.10.2-git.096b61a                armhf        libshairplay for XBian (armhf/master)
ii  xbian-package-splash              2.4.4                             armhf        Splash screen (rpi/master)
ii  xbian-package-transmission        20180331-0                        armhf        Metapackage for Transmission, configured for XBian
rc  xbian-package-tvheadend           4.2.8-31-1569379093               armhf        TVheadend (armhf/4.2)
ii  xbian-package-upstart-xbmc-bridge 2.0.2                             armhf        Bridge between xbmc event and upstart
ii  xbian-package-usbmount            2.2.1                             all          automatically mount and unmount USB mass storage devices
ii  xbian-package-vnc-server          2.5.4-1540479258                  armhf        VNC Server (rpi2/source-2.x)
ii  xbian-package-xbianhome           1.3.0-1570282452                  armhf        default user xbian home dir
ii  xbian-package-xbmc                18.5-1574268859                   armhf        XBMC (rpi2/leia)
ii  xbian-package-xbmc-scripts        1.9.4                             armhf        XBian XBMC upstart scripts.
ii  xbian-package-zram-swap           1.0.7                             armhf        zram-swap custom xbian package
ii  xbian-update                      20200109-0                        armhf        Updater package to XBian 20200109-0
```

mkreisl, 2021-03-25:
What are the kodi packages doing on the system?
It looks to me like an absolute mess installation
And Stretch has not been supported for over a year. So this is not going to work

Install a new image

closed 2021-03-25 by mkreisl

LaszloLango, 2021-03-26:
>What are the kodi packages doing on the system?

I did not istall them, the apt upgrade did it I guess. I do not want to lost my settings and database. Is it possible the backup of my home dir will reinstall the unnecessary packages.

mkreisl, 2021-03-26:
A backup of the home directory does not install any further packages, only the /home is completely backed up and restored if desired. 
This has nothing at all to do with the package manager
And if you have any additional packages installed, they will not be included in the /home backup either.