xbianonpi/xbian #894: kodi-gbm missing after updates

joebroesel, 2021-05-04:
Hello,

after the unattended upgrade on 2.5.21, my RPi4 booted onto the command line, not the GUI. Manual start of kodi on the command line did not work. Went back to snapshot before update -> Everything worked -> Unattended upgrades started again -> Same error again.

After some log file reading, it turned out that kodi-gbm is missing in /usr/local/lib/kodi/ 
It is still linked from /usr/local/lib/kodi/kodi.bin 

xbmc.log therefore shows:
May 04 09:21:50: start-stop-daemon: unable to stat /usr/local/lib/kodi/kodi.bin (No such file or directory)
start: Job failed to start
stop: Job has already been stopped: xbmc

I checked the updates that ran, and there is a list of packages that have been updated/removed but none of them led to errors and all show clean dpkg messages. Also can't see any other errors from the update:

Update: xbian-package-xbmc libcdio18

Removal: libcdio19 libdrm-amdgpu1 libdrm-common libdrm-nouveau2 libdrm-radeon1 libdrm2 libegl-mesa0 libegl1 libgbm1 libgl1-mesa-dri libglapi-mesa libgles2 libglvnd0 libimobiledevice6 libllvm9 libsensors-config libsensors5 libupower-glib3 libusbmuxd4 libwayland-client0 libwayland-server0 libxcb-dri2-0 libxcb-dri3-0 libxcb-present0 libxcb-sync1 libxcb-xfixes0 libxshmfence1 upower

Any ideas why kodi-gbm which seems to be the culprit got removed and by which package?
Any help appreciated!

mkreisl, 2021-05-05:
Sorry for the late response:

Please read through following post, you probably have the same mysterious problem.
http://forum.xbian.org/thread-4123-post-37039.htm

Something was wrong with the last xbian-packge-xbmc, no idea what was going on, it didn't cause any problems for me.

joebroesel, 2021-05-07:
Hi! 
Sorry as well, took some time until I could try. Just installed xbian-packge-xbmc (19.0-1620090104) from the staging area. Everything seems ok again and it boots into GUI.

Thanks for the quick fix!

closed 2021-05-07 by joebroesel