xbianonpi/xbian #895: Widevine 4.10.2252.0 not work with kodi on raspberry pi

jon10220, 2021-05-23:
Hello
does this problem concern xban? 

https://github.com/xbmc/inputstream.adaptive/issues/678

Does this mean netflix and other DRM plugins will stop working on xbian after 5/31/2021?
will there be any patch?

mkreisl, 2021-05-23:
> Hello
> does this problem concern xban?

Of course it also concerns XBian

2 things have to be done, see also https://github.com/LibreELEC/LibreELEC.tv/pull/5376

First, the glibc must be patched and rebuilt, because Debian does not do that - not even with Bullseye. This is of course not nice because the glibc is the central part of Linux and you might have a problem updating in the future. The patched glibc is already built at the moment.

Secondly the tcmalloc_minimal library (libtcmalloc_minimal.so) has to be built and preloaded. But this is no problem and is already running in test and obviously has no negative impact at the moment

jon10220, 2021-05-23:
Thank you for your response

I keep my fingers crossed for this to be done ...

by the way you do a great job thanks a lot!

vpeter4, 2021-06-07:
Btw in case missed: https://github.com/LibreELEC/LibreELEC.tv/pull/5422

mkreisl, 2021-06-07:
> Btw in case missed: [LibreELEC/LibreELEC.tv#5422](https://github.com/LibreELEC/LibreELEC.tv/pull/5422)

Can you please explain this in a little more detail?

Everything necessary has already been done. Kernel must be renewed and the patched libc6 from the staging repo are needed. Since libc6 is a central element, it has not been declared stable yet.

vpeter4, 2021-06-07:
Instead of preloading libtcmalloc or libwidevine we are using glibc patch to support 64 byte TLS memory alignment required for libwidevine. This way libwidevine library can be used with dlopen(). Seems preloading any of the above lib brings instability to Kodi even if Netflix or any other service using libwidevine is actually used.

mkreisl, 2021-06-07:
I have been using libtcmalloc_minimal.so preloading for weeks now and have not noticed any instabilities other than the usual ones

vpeter4, 2021-06-07:
I don't know what to say. With CoreELEC it's devs didn't noticed anything after long time of internal testing but users noticed stability issues right away.
At the end it is your choice which solution you will took - you have three options.

mkreisl, 2021-06-07:
Well, if this works more stable with the further libc6 patch, thenIi will do this and test it

thanks for the hint

matthuisman, 2021-06-10:
please post here what your solution is so I can update my blog post here:
https://www.matthuisman.nz/2021/05/widevine-arm-broken.html

mkreisl, 2021-06-11:
@matthuisman 
The two glibc patches are applied and the glibc has been rebuilt.
These packages are currently in the staging repository and will be moved to stable soon.

Just update the glibc, no further action is required

eikel, 2021-07-09:
I have updated the package sources to use XBian staging and upgraded libc6 to version 2.28-xbian1. Still the playback with Widevine fails. All packages are up-to-date, `apt` does not want to upgrade anything. System information:
```
xbian@raspberry4 ~ $ uname -a
Linux raspberry4 5.10.44+ #1 SMP PREEMPT Wed Jun 23 12:16:00 CEST 2021 armv7l GNU/Linux
```

This is a message from `kodi.log` containing hopefully all the relevant versions (sorry, it is in German, but the version numbers should still be extractable):
```
2021-07-09 19:39:01.307 T:5085  WARNING <general>: [script.module.inputstreamhelper]                                                                                                                                                                                                         
                                                   Kodi Version 19.1 läuft auf einem Linux System mit arm Architektur.            
                                                                
                                                   InputStream Helper ist in Version 0.5.7+matrix.1 
                                                   InputStream Adaptive ist in Version 2.6.17                                                                                                                                                                                                
                                                   
                                                   Widevine CDM ist in Version 4.10.2252.0 und wurde installiert am 2021-07-09 18:58
                                                   Es wurde extrahiert aus Chrome OS Image FIEVEL mit Version 13904.55.0
                                                   Zuletzt am 2021-07-06 19:06 auf Aktualisierungen geprüft
                                                   Es ist installiert in /home/xbian/.kodi/cdm/
                                                    
                                                   Bitte melden Sie Fehler hier: https://git.io/JfKJb
```

This is the part from the log showing the failure:
```
2021-07-09 19:39:39.802 T:5089     INFO <general>: initializing python engine.
2021-07-09 19:39:40.543 T:5089    FATAL <general>: [script.module.inputstreamhelper] ['ldd', '/home/xbian/.kodi/cdm/libwidevinecdm.so'] cmd failed.
2021-07-09 19:39:40.543 T:5089    FATAL <general>: [script.module.inputstreamhelper] Failed to check for missing Widevine libraries.
2021-07-09 19:39:40.544 T:5089  WARNING <general>: [script.module.inputstreamhelper] Widevine update check was made on 2021-07-06 19:06
2021-07-09 19:39:40.545 T:5089     INFO <general>: CPythonInvoker(24, /home/xbian/.kodi/addons/plugin.video.netflix/addon.py): script successfully run
2021-07-09 19:39:40.560 T:2832     INFO <general>: VideoPlayer::OpenFile: plugin://plugin.video.netflix/play/show/81166747/season/81166748/episode/81217432/?profile_guid=KWINUWE5ONFUZCIAJLODSVXQ4Q
2021-07-09 19:39:40.562 T:5123     INFO <general>: Creating InputStream 
2021-07-09 19:39:40.563 T:5123     INFO <general>: AddOnLog: inputstream.adaptive: SetVideoResolution (1920 x 1080)
2021-07-09 19:39:40.959 T:5123     INFO <general>: AddOnLog: inputstream.adaptive: Successfully parsed manifest file. #Periods: 1, #Streams in first period: 17, Type: VOD, Download speed: 2554490.5149 Bytes/s
2021-07-09 19:39:40.961 T:5123    ERROR <general>: AddOnLog: inputstream.adaptive: Unable to load widevine shared library (/home/xbian/.kodi/cdm/libwidevinecdm.so)
2021-07-09 19:39:40.961 T:5123    ERROR <general>: AddOnLog: inputstream.adaptive: OpenDRMSystem failed
2021-07-09 19:39:40.963 T:5123    ERROR <general>: CVideoPlayer::OpenInputStream - error opening [plugin://plugin.video.netflix/play/show/81166747/season/81166748/episode/81217432/?profile_guid=KWINUWE5ONFUZCIAJLODSVXQ4Q]
2021-07-09 19:39:40.963 T:5123     INFO <general>: CVideoPlayer::OnExit()
2021-07-09 19:39:40.964 T:5123     INFO <general>: ADDON: Dll Destroyed - InputStream Adaptive
2021-07-09 19:39:40.969 T:5086     INFO <general>: Deleting settings information for files plugin://plugin.video.netflix/play/show/81166747/season/81166748/episode/81217432/?profile_guid=KWINUWE5ONFUZCIAJLODSVXQ4Q
2021-07-09 19:39:40.984 T:2832     INFO <general>: CVideoPlayer::CloseFile()
2021-07-09 19:39:40.984 T:2832     INFO <general>: VideoPlayer: waiting for threads to exit
2021-07-09 19:39:40.984 T:2832     INFO <general>: VideoPlayer: finished waiting
2021-07-09 19:39:42.948 T:2832  WARNING <general>: OnMessage: Suspiciously long time to handle GUI_MSG_PLAYBACK_STARTED (2.35s)
```
`ldd` gives an error message:
```
xbian@raspberry4 ~ $ ldd /home/xbian/.kodi/cdm/libwidevinecdm.so
        not a dynamic executable
```
The file is parsable with `readelf`:
```

xbian@raspberry4 ~ $ readelf -d /home/xbian/.kodi/cdm/libwidevinecdm.so

Dynamic section at offset 0x816c58 contains 34 entries:
  Tag        Type                         Name/Value
 0x00000001 (NEEDED)                     Shared library: [libdl.so.2]
 0x00000001 (NEEDED)                     Shared library: [libpthread.so.0]
 0x00000001 (NEEDED)                     Shared library: [librt.so.1]
 0x00000001 (NEEDED)                     Shared library: [libm.so.6]
 0x00000001 (NEEDED)                     Shared library: [libc.so.6]
 0x00000001 (NEEDED)                     Shared library: [ld-linux-armhf.so.3]
 0x0000000e (SONAME)                     Library soname: [libwidevinecdm.so]
[... cut ...]
 0x0000000c (INIT)                       0x80e674
 0x0000000d (FINI)                       0x80e680
 0x6ffffff0 (VERSYM)                     0xc08
 0x6ffffffe (VERNEED)                    0xd54
 0x6fffffff (VERNEEDNUM)                 6
 0x00000000 (NULL)                       0x0
```

I also installed the Debian package libgoogle-perftools-dev and tried to start Kodi with  `LD_PRELOAD=/usr/lib/arm-linux-gnueabihf/libtcmalloc_minimal.so`. But that also did not work.

What did you do to get it running (with/without preloading)?

mkreisl, 2021-07-09:
> xbian@raspberry4 ~ $ ldd /home/xbian/.kodi/cdm/libwidevinecdm.so
>         not a dynamic executable

^^I would say that is your problem ^^

This is my output:
```
xbian@kmxbilr3 ~ $ ldd /home/xbian/.kodi/cdm/libwidevinecdm.so
        linux-vdso.so.1 (0xbeb5e000)
        /lib/libarmmem.so (0xb685d000)
        libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0xb6826000)
        libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0xb67be000)
        libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0xb67ab000)
        librt.so.1 => /lib/arm-linux-gnueabihf/librt.so.1 (0xb6795000)
        libnss3.so => /usr/lib/arm-linux-gnueabihf/libnss3.so (0xb66bf000)
        libnssutil3.so => /usr/lib/arm-linux-gnueabihf/libnssutil3.so (0xb6691000)
        libnspr4.so => /usr/lib/arm-linux-gnueabihf/libnspr4.so (0xb665c000)
        libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0xb655d000)
        /lib/ld-linux-armhf.so.3 (0xb6fbb000)
        libplc4.so => /usr/lib/arm-linux-gnueabihf/libplc4.so (0xb654a000)
        libplds4.so => /usr/lib/arm-linux-gnueabihf/libplds4.so (0xb6537000)
xbian@kmxbilr3 ~ $ ls -la /home/xbian/.kodi/cdm/libwidevinecdm.so
-rwxr--r-- 2 xbian xbian 7297324 22. Jun 18:47 /home/xbian/.kodi/cdm/libwidevinecdm.so
xbian@kmxbilr3 ~ $ 

```

eikel, 2021-07-09:
Thanks for your quick answer. I will try and see if there is another version of the shared object (maybe in some backups).

For the record: I used InputStream Helper inside Kodi and when it asked to install an old version, I selected “No” (also tried “Yes” without success). I also tried to restore a Widevine version using it and selected 4.10.2252.0. This version is also reported by InputStream Helper itself. Probably it is a problem with that plugin.

matthuisman, 2021-07-09:
https://k.slyguy.xyz/.decryptmodules/widevine/4.10.2252.0-linux-armv7.so

Or older version should also still work
https://k.slyguy.xyz/.decryptmodules/widevine/4.10.1679.0-linux-armv7.so

mkreisl, 2021-07-09:
I checked again which version I had just installed and was surprised to discover that I still had the old 4.10.1679.0 on this Raspberry Pi (Pi3).
I immediately used the current 4.10.2252.0, which also plays back Netflix without any problems.

eikel, 2021-07-10:
> https://k.slyguy.xyz/.decryptmodules/widevine/4.10.2252.0-linux-armv7.so

I had a look at this file and it gives me the same error message as the file inputstream.helper installed to my system:
```
xbian@raspberry4 /tmp $ ldd 4.10.2252.0-linux-armv7.so 
        not a dynamic executable
```

> 
> Or older version should also still work
> https://k.slyguy.xyz/.decryptmodules/widevine/4.10.1679.0-linux-armv7.so

If I understood xbmc/inputstream.adaptive#678 correctly, I thought that these old versions stopped working beginning on 2021-06-01. I tried to use it (also before with inputstream.helper), but playback does not start and the loading circle animation in the GUI is shown until I reboot the system.

mkreisl, 2021-07-10:
Your problems have really puzzled me
I just checked again on an XBian still running Debian Buster (all my other systems are already on Debian Bullseye for testing) and found that obviously the 4.10.2252.0 version is no longer usable under Buster, but the previous version seems to still work correctly.

```
root@kmxbilr2 /home/xbian/.kodi/cdm # ldd 4.10.1679.0-linux-armv7.so
        linux-vdso.so.1 (0xbec93000)
        /opt/lib/libmediaclient.so (0xb6798000)
        /lib/libarmmem.so (0xb6783000)
        libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0xb6744000)
        libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0xb66c9000)
        libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0xb66b6000)
        librt.so.1 => /lib/arm-linux-gnueabihf/librt.so.1 (0xb66a0000)
        libnss3.so => /usr/lib/arm-linux-gnueabihf/libnss3.so (0xb65d8000)
        libnssutil3.so => /usr/lib/arm-linux-gnueabihf/libnssutil3.so (0xb65ab000)
        libnspr4.so => /usr/lib/arm-linux-gnueabihf/libnspr4.so (0xb656b000)
        libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0xb6471000)
        /lib/ld-linux-armhf.so.3 (0xb6ef9000)
        libgcc_s.so.1 => /lib/arm-linux-gnueabihf/libgcc_s.so.1 (0xb6448000)
        libplc4.so => /usr/lib/arm-linux-gnueabihf/libplc4.so (0xb6434000)
        libplds4.so => /usr/lib/arm-linux-gnueabihf/libplds4.so (0xb6421000)
root@kmxbilr2 /home/xbian/.kodi/cdm # ldd 4.10.2252.0-linux-armv7.so
        das Programm ist nicht dynamisch gelinkt
root@kmxbilr2 /home/xbian/.kodi/cdm # 
```

The previous version also works fine, as I have just tested.
Please reinstall the previous version. The upgrade to Debian Bullseye will not be long in coming.

```
root@kmxbilr2 /home/xbian/.kodi/cdm # ll libwidevinecdm.so 
lrwxrwxrwx 1 xbian xbian 26 Jul 10 22:29 libwidevinecdm.so -> 4.10.1679.0-linux-armv7.so
root@kmxbilr2 /home/xbian/.kodi/cdm 
```

matthuisman, 2021-07-10:
Google havn't yet revoked the older versions on arm. I suspect due to how late some of the chrome os updates came out they are giving some time for everyone to update

mkreisl, 2021-07-10:
I have also suspected something like this

eikel, 2021-07-12:
After resetting my inputstream.adaptive settings back to standard, the old version also works for me. Thank you very much for your support!

jon10220, 2021-08-14:
Hello
I also tested the libc path from staging repo on a raspberry pi 4 and it doesn't work unfortunately.
on Widevine 4.10.2252.0 netflix addon does not work. 

https://pastebin.com/raw/idKxC46v

czy jest jakis inny sposób na uruchomienie Widevine 4.10.2252.0 w xbian?

mkreisl, 2021-08-16:
@jon10220, please read my earlier posts in this thread
It looks like the newest version of the library doesn't work under Debian Buster anymore, but the previous version still works fine, just checked again on Pi3 and Pi4
On Debian Bullseye, which will be used by default from the end of the week, the latest library runs fine, I use it myself on a Pi3 and Pi4.

jon10220, 2021-08-16:
Debian Bullseye will be ready for xbian in end of the week?
It's impressive! 
I read on the OSMC website (which is probably a similar operating system) that it could take them months or even a year... 

Anyway that's great news ... thanks a lot for keeping xbian alive, it's a great OS

mkreisl, 2021-08-16:
Yes, end of this week

BTW, I have been using Bullseye for XBian for months on a test basis.

matthuisman, 2021-08-16:
@mkreisl 
Are you using any patches in Bullseye to make the latest Widevine work?
I assume the older WV also still works as well

mkreisl, 2021-08-16:
@matthuisman, how many times do I have to mention this. Both the Buster and the Bullseye glibc have been patched in the same way. Buster obviously doesn't work properly anymore, but Bullseye does. And old Widevine lib still work unter Buster

matthuisman, 2021-08-16:
**your not mentioning if its you patching it or if its upstream.**

"Buster obviously doesn't work properly"
do you know why not if it also has the patches?

mkreisl, 2021-08-16:
I am only talking about patched glibc

I have not tested this further because 
a) it works under Bullseye and 
b) the old Widevine libraries are still working

To be honest I don't have the time to figure out why it doesn't work under Buster since Buster is practically history anyway

jon10220, 2021-08-22:
I checked for updates on my xbian ... and nothing new. Does the xbian Debian Bullseye version need to be installed from scratch?

mkreisl, 2021-08-22:
> I checked for updates on my xbian ... and nothing new. Does the xbian Debian Bullseye version need to be installed from scratch?

Currently only the newly created images are based on Bullseye. 
I have not yet enabled the updater from Buster to Bullseye, that will come later. In my experience, only a few people do this anyway, because the whole process takes a long time.

jon10220, 2021-08-22:
Thank you for your answer

I understand ... for me the installation via apt-get update is better, I don't have time to configure the media center from scratch.
If you put Bullseye in the staging repo for now, I'd love to test it

mkreisl, 2021-08-22:
Regardless of your answer, I put the required package **xbian-package-repo** in the staging repo.
This should be available on all mirrors in about 3h. After installing this, you can upgrade to Bullseye via the XBian Config GUI -> Update

jon10220, 2021-08-23:
The update was successful. On my raspberry pi 4b (CPU overclocked to 2GHZ) it took 70 minutes.
OS works fine including new Widevine 4.10.2252.0
Everything looks fine, thanks a lot ...

mkreisl, 2021-08-23:
Thanks for the feedback
Then I can close this issue

closed 2021-08-23 by mkreisl

matthuisman, 2021-09-01:
FYI:
looks like the old Widevine version has just been revoked.

So, for Xbian users - they just need to go to **XBian Config GUI > Update to Bullseye?**
i dont have xbian anywhere locally to write out more details instructions.

Is Xbian config just a kodi add-on? or is it cmdline?

mkreisl, 2021-09-01:
> Is Xbian config just a kodi add-on? or is it cmdline?

It is both. However the distribution upgrade function is only included in the Kodi GUI version

Then it's time to release the upgrade from Buster to Bullseye for everyone, because currently the required package (xbian-package-repo) is only available in the staging repository. Currently only new images are based on Bullseye

matthuisman, 2021-09-01:
Are you also able to do this?
https://github.com/LibreELEC/LibreELEC.tv/pull/5455/files
(It not done already)

It allows inputstream helper to know when the os supports the new widevine lib

mkreisl, 2021-09-02:
OMG, what a fiddling. Hopefully this will be removed from the inputstream.helper soon again

The environment variable has not been set yet of course

mkreisl, 2021-09-02:
@matthuisman 
so, I decided to use the 'ldd --version' variant. I couldn't get used to this disgusting workaround via the environment variable.

Btw, found an installation here which still works with libwidevinecdm.so version 4.10.1679.0
I was just surprised when I tested the patch detection via ldd it told me that there is an update available. But after aborting that hint it worked with the old version too.

... and, many thanks that you keep me up2date :smile:

matthuisman, 2021-09-02:
Only some add-ons still work. Netflix and Disney. But probably for only a
small amount of time.

mkreisl, 2021-09-02:
> Also,.your saying the below code will correctly detect support on xbian?

Yes, it does. But it was a bit tricky to make it work

```
root@kmxbilr2 /etc/init # ldd --version
ldd (Debian GLIBC 2.31-xbian1) 2.31 with arm64tls
Copyright © 2020 Free Software Foundation, Inc.
Dies ist freie Software; in den Quellen befinden sich die Lizenzbedingungen.
Es gibt KEINERLEI Garantie; nicht einmal für die TAUGLICHKEIT oder
VERWENDBARKEIT FÜR EINEN ANGEGEBENEN ZWECK.
Implementiert von Roland McGrath und Ulrich Drepper.
root@kmxbilr2 /etc/init 
```

matthuisman, 2021-09-02:
Ok thanks.
I'll add that to my widevine detection.
(My add-ons use different code to IA Helper)

mkreisl, 2021-09-02:
> Only some add-ons still work. Netflix and Disney. But probably for only a small amount of time.

I only have Netflix to test. But anyway, hopefully everything is ready now :smile:

matthuisman, 2021-09-02:
Sweet. So xbian all sorted..
I'll update my blog.

What's the instructions for existing users?
There an easy way to update?

mkreisl, 2021-09-02:
If you mean the upgrade to Bullseye, you will get a message box that a distribution upgrade is pending. This can then be easily started in the Kodi XBian-config GUI under Update.

And if you mean only the glibc, then it works the same way in the update GUI, there only the packages are renewed as usual with Debian.

matthuisman, 2021-09-02:
i dont use xbian so not familiar with that.

Just pretend im 60 year old man at home and my nephew installed Xbian for me and I use it to watch Coronations street which requires Widevine. Whats the most simple way for me to get it working?
Or is it best I just direct them all to your forums for advice?

mkreisl, 2021-09-02:
I would say let your nephew set it up :Smile:
Basically, these are just Kodi / Addon basis as everywhere else also

matthuisman, 2021-09-02:
Ok. I'm just going to send people to your forums

mkreisl, 2021-09-03:
that is what I said

horstle, 2021-09-09:
> OMG, what a fiddling. Hopefully this will be removed from the inputstream.helper soon again

Inputstreamhelper v0.5.8 removed the check. It's already in the Matrix repo, but not yet in Krypton/Leia. So you can probably remove the version string change in a few weeks.

mkreisl, 2021-09-10:
@horstle  oh I see it. Yesterday there was also an update. But never mind, the version string does not bother any further
But thanks for the hint