xbianonpi/xbian #898: Allow non root access to GPIO through /dev/gpiomem

tbertels, 2021-08-30:
Since the raspbian Jessie release the pi user is a member of the group "gpio" and so control of the GPIO no longer requires a change to the root user.
/dev/gpiomem has been created to allow non root access to gpio while keeping /dev/mem protected.

To fix this, xbian user should be added to a new gpio group and access to /dev/gpiomem should be allowed to the gpio group.

See https://github.com/RPi-Distro/pi-gen/blob/master/stage2/01-sys-tweaks/01-run.sh#L51
https://github.com/RPi-Distro/raspberrypi-sys-mods/blob/master/etc.arm/udev/rules.d/99-com.rules
https://github.com/eq-3/RaspberryMatic/blob/master/linux-4.1/drivers/char/broadcom/bcm2835-gpiomem.c

mkreisl, 2021-08-30:
Unfortunately this modification collides with the packages python3-rpi.gpio and rpi.gpio-common, which expect dialout for /sys/class/gpio and /dev/gpiomem as group
So if someone installs these packages as well, there will be chaos

tbertels, 2021-08-30:
In that case, is it possible to just add xbian to the dialout group?

mkreisl, 2021-08-30:
Of course it goes
I just had a look at Pi-OS how it is there. There everything is set to group gpio, because the associated udev rule is executed later than the one in the package rpi.gpio-common:armhf (60- vs 90-)
What a mess!

mkreisl, 2021-08-31:
The modifications have been done. It concerns the packages **xbian-update** and **xbian-package-xbianhome**
These are now available for testing in the **staging** repository.

Please install these packages and test if everything works. Thanks a lot

tbertels, 2021-08-31:
It works fine, thanks!

I see that you went with the gpio group in the end. It makes sense, and I wonder why the Debian version of rpi.gpio packages replaces gpio by dialout.

closed 2021-08-31 by tbertels