xbianonpi/xbian #899: SMB2/3 support

tbertels, 2021-08-30:
xbian currently enables by default SMB1.
SMB1 is disabled by default in Windows 10, so xbian doesn't show up in Windows "Network".

Fortunately, a daemon already exists for that: https://github.com/christgau/wsdd
When running, the script at https://github.com/christgau/wsdd/blob/master/src/wsdd.py immediately makes xbian visible on Windows 10 without installing SMB1.

Adding it to xbian services and enabling it by default would fix this.

mkreisl, 2021-09-01:
I have now integrated the **wsdd** daemon for testing, It is started together with the Samba nmbd daemon

Two wsdd parameters (hostname and workgroup) are taken from /etc/samba/smb.conf (server string and workgroup). Further parameters can be set in the configuration file /etc/wsdd.conf

It concerns the packages xbian-update and xbian-package-samba
These are now available for testing in the staging repository.

Please install these packages and test if everything work, unfortunately I don't have Windows 10 to test
Thanks a lot

tbertels, 2021-09-01:
I just tested it and it works fine after a reboot.
XBian shows up as expected in Windows 10 network browser.
Thank you!

closed 2021-09-01 by tbertels

tbertels, 2021-09-01:
I suggest commenting out the server string though as wsdd uses the hostname by default, unless the server string is defined in /etc/samba/smb.conf.

mkreisl, 2021-09-02:
This is exactly how it works. only if the server string is defined in smb.conf, it is also used for wsdd. If a user does not want this, he can comment it out in smb.conf.

tbertels, 2021-09-02:
Yes, and since the hostname can be easily changed in xbian-config, it would be easier to comment it out by default. That way, the user would just have to change the hostname with xbian-config. If the user has to find out about smb.conf and the server string and edit it anyway, it's pretty much useless that wsdd uses the hostname by default. By the way, smb.conf begins by "DO NOT EDIT THIS FILE".

mkreisl, 2021-09-02:
I thought you'd argue that way. However, I'm more concerned with compatibility, and so far XBian has been visible on the network under that name, and that's how it should stay. And so far no one has been bothered, so it probably won't be in the future. 
By the way, I usually have Samba disabled on my XBian installations anyway because I never need it.

tbertels, 2021-09-02:
I wonder if Samba itself uses the hostname by default when the server string isn't set. Given that the default hostname on XBian is XBian, it would also work.

mkreisl, 2021-09-02:
Nope, man page tells:

Default: server string = Samba %v

tbertels, 2021-09-02:
Ok, so that's where it should be fixed.

mkreisl, 2021-09-02:
No, Samba should integrate the wsdd. but they are reluctant to do so.