xbianonpi/xbian #900: Raspberry PI 3B with the latest XBian system, version with Debian bullseye - no wi-fi card.

zalinek, 2021-09-25:
I installed the XBian system on my Raspberry PI 3B from the newest, currently available image: XBian_Latest_rpi4.img.gz for Raspberry versions: Pi 2, Pi 3, Pi3B + and Pi 4B. It follows that my Raspberry should work fine with this system. However, I tried twice to cleanly install XBian from this image and each time during the initial setup, the wizard does not detect my wi-fi card in raspberry. There is only eth0 wired available, and that's it. I even went even further and in the terminal with the command: "sudo xbian-config network list" I checked the list of available network devices and it only gives me "eth0" in response. The wi-fi card in my raspberry is definitely working, because on other sd cards I have libreelec and osmc systems installed and there the wi-fi is detected and works properly. However, I didn't give up and decided to install XBian from a different, older image. I installed an older image from 07/30/2021. and wi-fi came out! I did all the available updates inside xbian and the wi-fi was still there, but there was one more big update left - a new Debian Bullseye distribution and after installing this bullseye distribution my wi-fi card is not available. It follows that the problem is inside the Debian Bullseye distribution. I decided not to install this update temporarily (bullseye), but I found out that this distribution is necessary for the latest Widevine version 4.10.2252.0 to work properly. The latest Widevine, on the other hand, is required for some streaming plug-ins to work properly. That's why I use the newest XBian on my raspberry so far, but I only use a network card - wired eth0. Please, investigate my problem with the XBian Bullseye for the wi-fi card in the Raspberry PI 3B. I know from friends that the latest XBian works on Raspberry PI 4B, but I only have RPI 3B so far. I really care about the latest XBian system because it has the new KODI version 19, a full (not truncated) Linux distribution with support for the latest Widevine, and it works on old analog PAL and NTSC TVs. Thank you in advance for your help in solving my problem and best regards ...

mkreisl, 2021-09-26:
The good news is, it is reproducible ... the even better news is that it is solvable

Please uninstall the package firmware-brcm80211 with the command `sudo apt-get purge firmware-brcm80211`

After a reboot wlan0 will be present. However, I have not yet checked what side effects the removal of the package might have.  In any case it looks like the package is broken at Bullseye

zalinek, 2021-09-27:
Thank you very much for your quick response. I uninstalled the indicated package: firmware-brcm80211 and wlan0 wi-fi now works without problem. I will continue to test XBian, introduce my settings, install the necessary plugins and if I encounter a problem, I will write it. Thank you again for your help and best regards.

mkreisl, 2021-09-29:
Just to close the matter here once and for all:

I have now gotten to the bottom of what the real cause of the problem is. 
The solution is to update the firmware and include the BLOB file in the kernel package.
See also the above commit

zalinek, 2021-09-29:
Excuse me but I don't understand what should I do now? Please unsubscribe in points:

1). .....
2). .....
3). .... etc.

complete commands that I execute sequentially in the terminal?
So far I have done this:

sudo apt-get purge firmware-brcm80211

and after a reboot, wi-fi works fine.

zalinek, 2021-09-29:
Or maybe it is enough to install the latest update that is available from today in XBian: "linux-image-bcm2836"?

mkreisl, 2021-09-29:
You do not need to do anything. you have already removed the firmware package.

zalinek, 2021-09-30:
Thank you, I understand now. Just one more question: should I install the latest update in XBian: "linux-image-bcm2836"?

mkreisl, 2021-09-30:
You can do this as with every update. Do it or leave it. Since everything works for you, this is not absolutely necessary.