xbianonpi/xbian #901: soundcard Chipdip DAC (dsp machine) not included in new release.

chipdipru, 2021-10-02:
Please add the card driver snd-soc-chipdip-dac.ko it is not included in the new release.
[image]

Overlay (chipdip-dac) does not work without a driver. It is not possible to turn on the sound card.
This is a popular sound card in Russia. 
ADAU1701
[image]

mkreisl, 2021-10-02:
Which driver?

I only find binary only modules from here: https://github.com/chipdipru
And including binary drivers is very suspect to me

chipdipru, 2021-10-02:
The driver is in the kernel https://github.com/raspberrypi/linux/blob/rpi-5.10.y/sound/soc/bcm/chipdip-dac.c
Overlay and driver were added in the kernel simultaneously. But it happened somehow that only overlay was taken into account without the driver.

mkreisl, 2021-10-02:
Ah, now it becomes clearer to me. I stupidly searched for the presence in the wrong kernel sources (5.4) and could not find anything

Next time the kernel package is built, the module is included

closed 2021-10-02 by mkreisl