xbianonpi/xbian #907: Raspberry PI 3B with the latest XBian - OnOff SHIM Pimoroni does not install...

zalinek, 2021-10-30:
I'm sorry for "curl", but I don't know linux well and I'm just learning. I didn't know that the curl service should be installed. Once again, I apologize and ask for more help as the problem is now more difficult. After entering the command:

curl https://get.pimoroni.com/onoffshim | bash

is the following system response:

xbian @ xbian ~ $ curl https://get.pimoroni.com/onoffshim | bash
  % Total% Received% Xferd Average Speed ​​Time Time Time Current
                                 Dload Upload Total Spent Left Speed
100 38 737 100 38 737 0 0 6925 0 0:00:05 0:00:05 -: -: - 9894

This script will install everything needed to use
On / Off SHIM

Always be careful when running scripts and commands copied
from the internet. Ensure they are from a trusted source.

If you want to see what this script does before running it,
you should run: 'curl https://get.pimoroni.com/onoffshim'

Your operating system is not supported, sorry!

xbian @ xbian ~ $ 

Thank You very much for all your help and best regards.

mkreisl, 2021-10-30:
Please read the last line of the output: ``` Your operating system is not supported, sorry! ```

I have adapted the script and it can also be installed.

**But attention**: because I do not have the necessary hardware, the Raspberry Pi shuts down immediately after a reboot.
so be sure what you are doing

```
curl  https://gist.githubusercontent.com/mkreisl/e84d455bea1908413bbf0ee9c73f5ec6/raw/4f5e811e254a2390c33bbd0ce6b803f588dc06f4/onoffshim | bash
```

For everything else to this topic no more support since no hardware available in this regard

zalinek, 2021-11-01:
Thank you, I installed the script, but the raspberry does not turn off completely - after pressing the onoff shim button, the system shuts down ok, but it does not turn off the power of the raspberry (the lights are on). Now, when I want to turn on the raspberry again, I have to disconnect it from the power supply first, wait a few seconds and reconnect the power cable and then I can turn on the raspberry again with the onoff shim button. I have a big request, could you please correct the script so that the raspberry turns off completely (the power was turned off - the lights are off)? Thank you in advance for all your help and best regards.

mkreisl, 2021-11-02:
> For everything else to this topic no more support since no hardware available in this regard

What is there not to understand about this sentence?

By the way, such a power button is complete nonsense, because the power consumption only decreases imperceptibly. this was only solved a bit more sensibly with a Pi4. But if I remember correctly you have a Pi3B+.

By the way, I have once the GPIO3 solution as described here (https://www.stoffl.info/2020/03/31/how-to-raspberry-pi-4-gpio-ein-und-ausschalten-power-on-off-button/) tested and it works on a Pi3 as well as on a Pi4 (assuming the right firmware is installed and configured).

If that still does not work, you have to contact the manufacturer (they have a forum), because this has nothing to do with XBian anymore