xbianonpi/xbian #908: Rpi4 - Xbian_2021.11.05_Rpi4.img - Kernel Panic-not syncing: VFS: Unable to mount root fs on unknown-block

slrslr, 2021-11-10:
I have loaded the img file to the SD card using Raspberry Pi Imager v1.6.2 which appeared to be verifying the disk after copied the data.

After putting disk to Raspberry Pi 4:

CPU: 3 PID: 0 Comm: swapper/3 Not Tainted
...
Kernel Panic-not syncing: VFS: Unable to mount root fs on unknown-block

[image]

Do you want to know any kind of hash of the image file, or such to verify integrity or what to try? I am on Linux currently.

mkreisl, 2021-11-10:
You don't do that with this imager to write the XBian image either!
See the raspberries at the top of the screen. These definitely do not exist in XBian
Whatever you wrote there on the sd-card is complete garbage

closed 2021-11-10 by mkreisl

slrslr, 2021-11-10:
@mkreisl on the https://xbian.org/getxbian/ & https://xbian.org/getting-started/ i am unable to find some proper way to install the img.gz file i have already downloaded if not using rpi-imager  And on the sourceforge page where is hosted your own installer is no information in description how to use it.
Can i do something like:
$ sudo dd if=your.img.gz of=/dev/MySDToOverwrite status="progress"
?

mkreisl, 2021-11-10:
$gunzip your.img.gz
$sudo dd if=your.img of=/dev/MySDToOverwrite status="progress"

slrslr, 2021-11-10:
Thanks @mkreisl , it appears there is some issue on my end. gunzip says: invalid compressed data--format violated
tar: This does not look like a tar archive

I do not trust the integrity of the image. When i copy it, it always has different md5 hash, maybe it is normal, i do not know. Is there any way You tell me correct hash or the way to verify the image Xbian_2021.11.05_Rpi4.img is the same as your source? Or too much work to supply hashes on the download page?

mkreisl, 2021-11-11:
I really don't know what you are doing. I have downloaded said image and the XBian_Latest_rpi4.img.gz and unpacked it properly. There is no problem. By the way, this is NOT a tar archive, this is just a zipped image file.
You should do some troubleshooting. If you always get a different md5 sum there is definitely something wrong with you.

By the way, the md5 sum of the above archive is 8ece591ea2a88f2085559a28cf829b84

```
kmnote5:/srv/scratch/Load/Unpacked/xbian # md5sum XBian_2021.11.05_rpi4.img.gz
8ece591ea2a88f2085559a28cf829b84  XBian_2021.11.05_rpi4.img.gz
kmnote5:/srv/scratch/Load/Unpacked/xbian # gunzip XBian_2021.11.05_rpi4.img.gz
gzip: XBian_2021.11.05_rpi4.img already exists; do you wish to overwrite (y or n)? y
kmnote5:/srv/scratch/Load/Unpacked/xbian # 
```

I think in this case hashes are totally unnecessary, because if the gz file is not correct then it can not be unpacked. And if someone can manipulate the gz file then he can manipulate the hashes too

slrslr, 2021-11-11:
My md5sum was different. So I have tried to download the file again (this time not to an unreliable USB SD card but to a HDD):
`wget --no-check-certificate --show-progress https://kumisystems.dl.sourceforge.net/project/xbian/release/XBian_Latest_rpi4.img.gz`
then again same command only with -c switch to continue/in case file is incomplete. Said "The file is already fully retrieved; nothing to do."
`md5sum XBian_Latest_rpi4.img.gz` shown same sum as yours
`gunzip XBian_Latest_rpi4.img.gz`
then i could try to copy resulting XBian_Latest_rpi4.img to SD card using Linux command `$sudo dd if=your.img of=/dev/MySDToOverwrite status="progress"`
But i have first wanted to try [rpi-imager app](https://github.com/raspberrypi/rpi-imager) again to see if it will work. So it copied .img to different SD card.
Now Rpi4 booted Xbian without issue.
So i think @mkreisl You were wrong when you have said "You don't do that with this imager to write the XBian image either!".

Thank you much for your help. **The issue was caused by loading corrupted image likely.**

mkreisl, 2021-11-11:
That's good news. Nevertheless I have included the hash generation (SHA512) and this will be available for download tomorrow parallel to the image files, although I still don't see much sense in it.