xbianonpi/xbian #909: Newbie feedback to developers (initial boot many codecs to click; unknown SSH credentials)

slrslr, 2021-11-11:
Hello, i have started Xbian_2021.11.05_Rpi4 on my Rpi4 today for the first time.

I wanted to tell developers what i was missing as a newbie user in order they consider improving.

1)
After boot i was surprised i had to confirm not to install various codecs etc. had to click maybe 30 times OK button.
That was not a problem, i have just not understood most of the things that was offered to me.
So it may help if user can select category of codecs to install after initial boot. Like three options (basic codes, basic tv codecs, encoding codecs..) or install all showing disk space needed next to each group.

2) Next thing i was looking for was to try to restore my .tar backup file created my LibreELEC. Maybe it is not possible, but it raised issue:
What is the SSH login credentials. This information was missing at https://xbian.org/getting-started/ . This information is visible at the bottom of the https://xbian.org/faq/ (Default login is xbian and password is raspberry).
When i have used these credentials and tried to upload file to /xbmc-backup & put_here_to_restore. There was network traffic, yet the file disappeared. I have not found option to allow write access and after some time i have found somewhere option to allow SSH root access in Settings/Xbian/System. I did it expecting the password will be the same "raspberry", no i could not login, had to go back and set password via UI: Settings/Xbian/System
Improvement? mention login credentials on [getting started page](https://xbian.org/getting-started/). Update FAQ entry to mention what is the root password.

Also can be handy to mention if/how one can restore LibreELEC settings like tvheadend and its configuration? But it is too tageted and many people not use it so maybe not even deserve FAQ.

mkreisl, 2021-11-11:
Just a few sentences to explain

1) XBian is not Libreelec
Libreelec is a collection of individual programs that admittedly worked well. But since LE does not have a package manager, everything has to be done via addons. This is not the case with XBian. XBian is based on Debian and therefore also has the package manager. 
2) What do you want with LE backup files at XBian????
3) the xbian-backup folder is for exchanging backup files created by xbian backup tools. nothing else.
4) as it is usual with debian/ubuntu etc etc, there is normally no root access. consequently no root password.

XBian also has a forum which is actually responsible for this. please avoid leaving such things as issues here in the future.

Thanks a lot

closed 2021-11-11 by mkreisl

slrslr, 2021-11-11:
1. so in another words you are saying it is impossible to prevent stupid (for basic user who does not understand codecs and such) tens of prompts i have mentioned?

2. Restore it. Are you that dumb? No, you probably do not know that this backup is regular .tar archive which has many common folders found on Xbian. Both have in comon to be running kodi and i expect addons and its settings are same. Also most of the Kodi settings are the same. Simply this was request for any developer who may have time to make this happen. And you have ignored my feedback regarding passwords.

3. I am sorry i have updated my post after You have started reading it so you have not read it or you have ignored my feedback regarding character set/language. Please and try to be less in hurry closing the issues.

4. Please read point 4 again, you have not addressed it. I am sorry again for updating my post, not expected so fast reading.