xbianonpi/xbian #910: xbian password need to change - even i have changed it

slrslr, 2021-11-13:
Initial startup cause window inviting to change default user password. I have changed it, then via SSH changed it again. And after each reboot there is this stupid invitation again and again even password "hash" is certainly different from default "raspberry".

closed 2021-11-13 by mkreisl

slrslr, 2021-11-13:
@mkreisl explanation, or just plain arrogance?

mkreisl, 2021-11-13:
You will not get any more support from me here. And you know why. This is not a help forum
I say this for the third time

slrslr, 2021-11-13:
@mkreisl I am not aware about that and that this is support request. It is possible bug report. I think that it is obvious. that this change password window should not appear if password was changed which means bug or idea for improvement. Obvious (for clever people)

mkreisl, 2021-11-13:
Most mistakes are made by the person sitting in front of the screen, as in this case again.

THIS IS NOT AN ISSUE

slrslr, 2021-11-15:
Option 2 **worked** for me to workaround this bug:

Option 1: Kodi XBian-Config GUI
Kodi->Settings->XBian->System->XBian password: change password and watch for the sucessfully Toast message

Option 2: SSH connection and xbian-config CLI
sudo xbian-config (if not already running)
2 Settings->2 Accounts->2 XBian password: change password and notice the successful messagebos

Option 3: (not recommended!): If you act too stupid or whatever:
sudo touch /usr/local/include/xbian-config/config/xbianpass