xbianonpi/xbian #912: Kodi freeze/crash on boot with 4k TV

tbertels, 2022-01-06:
When booting Xbian with rpi4b connected to a 4k TV, Kodi crashes (the screens stays stuck on the Matrix style background).
After reading  "4K Playback on rpi4" (http://forum.xbian.org/thread-4133-post-37278.html#pid37278),
I tried changing 

`dtoverlay=vc4-fkms-v3d,cma-384`
to
`dtoverlay=vc4-kms-v3d,cma-384`
in /boot/config.txt
and it works.
(I also had to change the audio output to S/PDIF in Kodi to have sound since that's what's being used here for now).

A few drawbacks:
* VNC server doesn't seem to work with kms
* The transition between video in fullscreen and video in background of the menu (while still playing) causes the sound to stop temporarily and the screen to turn black.
* The Kodi start animation isn't played anymore (the screen stays black or something then goes straight to the home menu)

I also tried with and without the staging firmware, but it made no difference. I'll report any problem I notice.

Despite the drawbacks, I just using kms by default for the rpi4, unless an option can be added to xbian-config to enable it (with a notice about 4k screen compatibility).

By the way, kms is needed for a 4k TV even if we use Kodi in 1920x1080.

tbertels, 2022-01-06:
Note that there may be issues with hdmi audio: https://github.com/raspberrypi/linux/issues/4543

mkreisl, 2022-01-06:
Thanks that finally someone has tested 4k resolution, I can not due to lack of hardware. I have already considered several times to switch from fkms to kms in new built images, but have not done it because of other small problems, which you also partly mentioned here.

For example, because VNC doesn't work anymore. This is a long known problem, for this I used the solution of lrusak and integrated the VNC server directly in Kodi (his version is not really usable), but only in the version 20. I use meanwhile therefore also already some weeks the kms driver

The animated Kodi start doesn't work either, because this still requires the old dispmanx functionality. An alternative solution with ffplay or the like is unfortunately not usable. Another solution might work, but I haven't got around to testing it yet.

The black screen and the audio dropout when switching to video background (in my opinion a pretty useless function anyway) is probably due to the fact that the screen mode has to be switched, and that just takes some time

tbertels, 2022-01-06:
It used to work until a few months ago on the 4k TV with fkms, then an update broke it. It may just be a problem with the Kodi config file, I'll try to reset it.

The video background is optional, but it's smoother with fkms because there isn't that screen mode change.
And the sound is cut too early with kms, which doesn't look polished (though it may only be so with S/PDIF).

mkreisl, 2022-01-17:
> It used to work until a few months ago on the 4k TV with fkms, then an update broke it.

This is probably due to the firmware, which also broke the PI:HDMI audio device at kms

We'll just have to live with that (for a while?), since everything happens in the kernel (opensource) and no longer via the firmware (closedsource) as with fkms.

Progress report:
The current xbian-package-xbmc 19.3 now includes the VNC server, 
and the splash video for the kms driver is coming soon.