xbianonpi/xbian #914: xbmc service stopped working, boot drops to terminal

Sjoerd82, 2022-01-20:
As per today, and having done no updates whatsoever, the xbmc service fails to start. Symptoms are:
* After boot, the screen waits on network for a longer than usual time, and then drops to the shell.
* When manually starting the xbmc service, the KODI splash screen starts, but eventually drops to shell before reaching the KODI interface.
* There are no current log entries anywhere, which may have something to do with that it seems the file system is readonly
* When SSH'ing into the device, there is this message: /usr/local/include/xbian-config/functions/language: line 23: /tmp/xbian-config.lang: Read-only file system
* "sudo btrfs scrub start /;sudo watch btrfs scrub status /" shows no errors
* suspecting there's something wrong with the SD-card I tried to do some diagnostics, however, I'm not familiar with btrfs and how to check it. I tried "sudo btrfs check --force /dev/mmcblk0p2" (without unmounting), and got:
Opening filesystem to check...
WARNING: filesystem mounted, continuing because of --force
Checking filesystem on /dev/mmcblk0p2
UUID: e44233d7-e6ac-4dd5-8853-58e5768144ff
[1/7] checking root items
[2/7] checking extents
Csum didn't match
Csum didn't match
owner ref check failed [72237056 16384]
owner ref check failed [149946368 16384]
ref mismatch on [1793175552 4096] extent item 1, found 0
incorrect local backref count on 1793175552 parent 149946368 owner 0 offset 0 fo                                                    und 0 wanted 1 back 0x1246490
backref disk bytenr does not match extent record, bytenr=1793175552, ref bytenr=                                                    0
backpointer mismatch on [1793175552 4096]
owner ref check failed [1793175552 4096]
ref mismatch on [1793830912 24576] extent item 1, found 0
incorrect local backref count on 1793830912 parent 149946368 owner 0 offset 0 fo                                                    und 0 wanted 1 back 0x1248060
backref disk bytenr does not match extent record, bytenr=1793830912, ref bytenr=                                                    0
backpointer mismatch on [1793830912 24576]
owner ref check failed [1793830912 24576]
ref mismatch on [1793855488 12288] extent item 1, found 0
incorrect local backref count on 1793855488 parent 149946368 owner 0 offset 0 fo                                                    und 0 wanted 1 back 0x1248160
backref disk bytenr does not match extent record, bytenr=1793855488, ref bytenr=                                                    0
backpointer mismatch on [1793855488 12288]
owner ref check failed [1793855488 12288]
ref mismatch on [1793867776 20480] extent item 1, found 0
incorrect local backref count on 1793867776 parent 149946368 owner 0 offset 0 fo                                                    und 0 wanted 1 back 0x1248260
backref disk bytenr does not match extent record, bytenr=1793867776, ref bytenr=                                                    0
backpointer mismatch on [1793867776 20480]
owner ref check failed [1793867776 20480]
ref mismatch on [1793888256 8192] extent item 1, found 0
incorrect local backref count on 1793888256 parent 149946368 owner 0 offset 0 fo                                                    und 0 wanted 1 back 0x1248360
backref disk bytenr does not match extent record, bytenr=1793888256, ref bytenr=                                                    0
backpointer mismatch on [1793888256 8192]
owner ref check failed [1793888256 8192]
ref mismatch on [1988202496 61440] extent item 1, found 0
incorrect local backref count on 1988202496 parent 149946368 owner 0 offset 0 fo                                                    und 0 wanted 1 back 0x13bb9f0
backref disk bytenr does not match extent record, bytenr=1988202496, ref bytenr=                                                    0
backpointer mismatch on [1988202496 61440]
owner ref check failed [1988202496 61440]
ref mismatch on [1988669440 45056] extent item 1, found 0
incorrect local backref count on 1988669440 parent 149946368 owner 0 offset 0 fo                                                    und 0 wanted 1 back 0x13bfd30
backref disk bytenr does not match extent record, bytenr=1988669440, ref bytenr=                                                    0
backpointer mismatch on [1988669440 45056]
owner ref check failed [1988669440 45056]
ref mismatch on [1995440128 188416] extent item 1, found 0
incorrect local backref count on 1995440128 parent 149946368 owner 0 offset 0 fo                                                    und 0 wanted 1 back 0x1bbe430
backref disk bytenr does not match extent record, bytenr=1995440128, ref bytenr=                                                    0
backpointer mismatch on [1995440128 188416]
owner ref check failed [1995440128 188416]
ERROR: errors found in extent allocation tree or chunk allocation
[3/7] checking free space cache
[4/7] checking fs roots

( ..to fast to copy.. )

root 2781 inode 95920 errors 2001, no inode item, link count wrong
        unresolved ref dir 44288 index 12 namelen 11 name __pycache__ filetype 2 errors 4, no inode ref
root 2781 inode 95921 errors 2001, no inode item, link count wrong
        unresolved ref dir 44291 index 52 namelen 11 name __pycache__ filetype 2 errors 4, no inode ref
ERROR: errors found in fs roots
found 3807178752 bytes used, error(s) found
total csum bytes: 3593668
total tree bytes: 95879168
total fs tree bytes: 86654976
total extent tree bytes: 4489216
btree space waste bytes: 22463612
file data blocks allocated: 4853329920
 referenced 4830142464

Now, this looks bad, or is it because the FS was mounted. Any assistance would be greatly appreciated! :-)

mkreisl, 2022-01-20:
1) enable initramfs in /boot/config.txt (remove # from already existing line `initramfs initramfs.gz followkernel`)
2) add rescue to /boot/cmdline.txt (should look like `rescue telnet zswap.enabled=1 ....`)
3) boot into rescue and run btrfs check from there

Then you can see if your file system is broken (which I assume is the case).

If you don't have an image backup, reinstall completely from a new image, because in my opinion it's not worth the effort to fix it.

Sjoerd82, 2022-01-20:
thx, will try that

Sjoerd82, 2022-01-21:
Yup, SD card was f******. RPi likes to eat those like they're going out of fashion :-(

closed 2022-01-21 by Sjoerd82

mkreisl, 2022-01-21:
That's why I switched all my development systems to sd-card-less operation years ago.