xbianonpi/xbian #915: rpi-eeprom on rpi4 fails creating /usr/local/bin/vcgencmd symlink to ./vcgencmd - with workaround

john-coleman, 2022-01-24:
Thanks for the distro, very smooth.

Latest updates as of 2022-01-24 22:08 UTC
Raspberry Pi4
```
root@pi4b1 ~ # rpi_revision
memory: 8192 MB
processor: Broadcom BCM2711
i2cDevice: /dev/i2c-1
model: Raspberry Pi 4 Model B
manufacturer: Sony UK
pcb revision: 4
warranty void: no
revision: d03114
peripheral base: 0xFE000000

checking peripheral base against bcm_host library
... peripheral base matches base from bcm_host
```

I have tried 13.3-1~xbian (latest available at time of writing) and 13.0.1~xbian3, same result.

```
xbian@pi4b1 ~/rpi-eeprom $ sudo apt install rpi-eeprom=13.0-1~xbian3
Reading package lists... Done
Building dependency tree... Done
  rpi-eeprom
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 934 kB of archives.
After this operation, 22.0 MB of additional disk space will be used.
Get:1 http://apt.xbian.org stable/main armhf rpi-eeprom armhf 13.0-1~xbian3 [934 kB]
Fetched 934 kB in 1s (708 kB/s)
Create a snapshot of '/run/xbian-storage/root/@' in '/run/xbian-storage/root/@btrfs-auto-snap_apt-run-2022-01-24-2137'
Delete subvolume (no-commit): '/run/xbian-storage/root/@last_good_known'
Create a snapshot of '/run/xbian-storage/root/@btrfs-auto-snap_apt-run-2022-01-24-2137' in '/run/xbian-storage/root/@last_good_known'
Delete subvolume (no-commit): '/run/xbian-storage/root/@btrfs-auto-snap_apt-run-2022-01-24-2129'
@btrfs-auto-snap_apt-run-2022-01-24-2137, 1 created snapshots, 0 sent snapshots, 1 destroyed snapshots, 0 created filesystems, 0 warnings.
Selecting previously unselected package rpi-eeprom.
(Reading database ... 46959 files and directories currently installed.)
Preparing to unpack .../rpi-eeprom_13.0-1~xbian3_armhf.deb ...
Unpacking rpi-eeprom (13.0-1~xbian3) ...
Setting up rpi-eeprom (13.0-1~xbian3) ...
ln: failed to create symbolic link './vcgencmd': File exists
dpkg: error processing package rpi-eeprom (--configure):
 installed rpi-eeprom package post-installation script subprocess returned error exit status 1
Processing triggers for man-db (2.9.4-2) ...
Errors were encountered while processing:
 rpi-eeprom
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

Doing a clean `sudo su -` (as xbian and root $PATH differ slightly, though that shouldn't matter to dpkg)
```
xbian@xbian ~/rpi-eeprom $ sudo su -
root@pi4b1 ~ # sudo apt remove rpi-eeprom
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages were automatically installed and are no longer required:
  libpci3 pci.ids pciutils
Use 'apt autoremove' to remove them.
The following packages will be REMOVED:
  rpi-eeprom
0 upgraded, 0 newly installed, 1 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 22.0 MB disk space will be freed.
Do you want to continue? [Y/n] y
Create a snapshot of '/run/xbian-storage/root/@' in '/run/xbian-storage/root/@btrfs-auto-snap_apt-run-2022-01-24-2151'
Delete subvolume (no-commit): '/run/xbian-storage/root/@last_good_known'
Create a snapshot of '/run/xbian-storage/root/@btrfs-auto-snap_apt-run-2022-01-24-2151' in '/run/xbian-storage/root/@last_good_known'
Delete subvolume (no-commit): '/run/xbian-storage/root/@btrfs-auto-snap_apt-run-2022-01-24-2137'
@btrfs-auto-snap_apt-run-2022-01-24-2151, 1 created snapshots, 0 sent snapshots, 1 destroyed snapshots, 0 created filesystems, 0 warnings.
(Reading database ... 47029 files and directories currently installed.)
Removing rpi-eeprom (13.0-1~xbian3) ...
Processing triggers for man-db (2.9.4-2) ...
root@pi4b1 ~ # apt install rpi-eeprom
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Recommended packages:
  flashrom
The following NEW packages will be installed:
  rpi-eeprom
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 1,089 kB of archives.
After this operation, 23.5 MB of additional disk space will be used.
Get:1 http://apt.xbian.org stable/main armhf rpi-eeprom armhf 13.3-1~xbian [1,089 kB]
Fetched 1,089 kB in 0s (2,891 kB/s)
Create a snapshot of '/run/xbian-storage/root/@' in '/run/xbian-storage/root/@btrfs-auto-snap_apt-run-2022-01-24-2152'
Delete subvolume (no-commit): '/run/xbian-storage/root/@last_good_known'
Create a snapshot of '/run/xbian-storage/root/@btrfs-auto-snap_apt-run-2022-01-24-2152' in '/run/xbian-storage/root/@last_good_known'
Delete subvolume (no-commit): '/run/xbian-storage/root/@btrfs-auto-snap_apt-run-2022-01-24-2138'
@btrfs-auto-snap_apt-run-2022-01-24-2152, 1 created snapshots, 0 sent snapshots, 1 destroyed snapshots, 0 created filesystems, 0 warnings.
Selecting previously unselected package rpi-eeprom.
(Reading database ... 46959 files and directories currently installed.)
Preparing to unpack .../rpi-eeprom_13.3-1~xbian_armhf.deb ...
Unpacking rpi-eeprom (13.3-1~xbian) ...
Setting up rpi-eeprom (13.3-1~xbian) ...
ln: failed to create symbolic link './vcgencmd': File exists
dpkg: error processing package rpi-eeprom (--configure):
 installed rpi-eeprom package post-installation script subprocess returned error exit status 1
Processing triggers for man-db (2.9.4-2) ...
Errors were encountered while processing:
 rpi-eeprom
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

Inspecting `/var/lib/dpkg/info/rpi-eeprom.postinst` I see
```
case "$1" in
    configure)
        [ -L /usr/local/bin/vcgencmd ] || ln -s $(which vcgencmd) /usr/local/bin/vcgencmd
    ;;

    abort-upgrade|abort-remove|abort-deconfigure)
    ;;

    *)
        echo "postinst called with unknown argument \`$1'" >&2
        exit 1
    ;;
esac
```

Applying manual work-around:

```
root@pi4b1 ~ # which vcgencmd
/opt/vc/bin/vcgencmd
root@pi4b1 ~ # ls -lh /usr/local/bin/vcgencmd
ls: cannot access '/usr/local/bin/vcgencmd': No such file or directory
root@pi4b1 ~ # ln -s $(which vcgencmd) /usr/local/bin/vcgencmd
root@pi4b1 ~ # which vcgencmd
/usr/local/bin/vcgencmd
root@pi4b1 ~ # apt install
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Create a snapshot of '/run/xbian-storage/root/@' in '/run/xbian-storage/root/@btrfs-auto-snap_apt-run-2022-01-24-2203'
Delete subvolume (no-commit): '/run/xbian-storage/root/@last_good_known'
Create a snapshot of '/run/xbian-storage/root/@btrfs-auto-snap_apt-run-2022-01-24-2203' in '/run/xbian-storage/root/@last_good_known'
Delete subvolume (no-commit): '/run/xbian-storage/root/@btrfs-auto-snap_apt-run-2022-01-24-2144'
@btrfs-auto-snap_apt-run-2022-01-24-2203, 1 created snapshots, 0 sent snapshots, 1 destroyed snapshots, 0 created filesystems, 0 warnings.
Setting up rpi-eeprom (13.3-1~xbian) ...
```

If I recall correctly, apt install runs the postinst scripts as root, but I presume with a different $PATH, which may explain why `$(which vcgencmd)` is returning `./vcgencmd` instead of `/opt/vc/bin/vcgencmd` during `postinst`.

I have not yet found where xbian is versioning these package builds, or if they're actually different from the mainline `rpi-eeprom`.

Hope this helps.

mkreisl, 2022-01-25:
Sorry, but I can't see any problem:

```
root@kmxbilr4 ~ # apt-get install rpi-eeprom
Paketlisten werden gelesen… Fertig
Abhängigkeitsbaum wird aufgebaut… Fertig
Statusinformationen werden eingelesen… Fertig
Die folgenden zusätzlichen Pakete werden installiert:
  libpci3 pci.ids pciutils
Empfohlene Pakete:
  flashrom
Die folgenden NEUEN Pakete werden installiert:
  libpci3 pci.ids pciutils rpi-eeprom
0 aktualisiert, 4 neu installiert, 0 zu entfernen und 5 nicht aktualisiert.
Es müssen 1.462 kB an Archiven heruntergeladen werden.
Nach dieser Operation werden 25,1 MB Plattenplatz zusätzlich benutzt.
Möchten Sie fortfahren? [J/n] 
Holen:1 http://kmcubie.kmhome.com stable/main armhf rpi-eeprom armhf 13.3-1~xbian [1.089 kB]
Holen:2 http://deb.debian.org/debian bullseye/main armhf pci.ids all 0.0~2021.02.08-1 [224 kB]
Holen:3 http://deb.debian.org/debian bullseye/main armhf libpci3 armhf 1:3.7.0-5 [57,1 kB]
Holen:4 http://deb.debian.org/debian bullseye/main armhf pciutils armhf 1:3.7.0-5 [91,9 kB]
Es wurden 1.462 kB in 1 s geholt (2.464 kB/s).
Create a snapshot of '/run/xbian-storage/root/@' in '/run/xbian-storage/root/@btrfs-auto-snap_apt-run-2022-01-25-2010'
Delete subvolume (no-commit): '/run/xbian-storage/root/@last_good_known'
Create a snapshot of '/run/xbian-storage/root/@btrfs-auto-snap_apt-run-2022-01-25-2010' in '/run/xbian-storage/root/@last_good_known'
Delete subvolume (no-commit): '/run/xbian-storage/root/@btrfs-auto-snap_apt-run-2022-01-15-1758'
@btrfs-auto-snap_apt-run-2022-01-25-2010, 1 created snapshots, 0 sent snapshots, 1 destroyed snapshots, 0 created filesystems, 0 warnings.
Vormals nicht ausgewähltes Paket pci.ids wird gewählt.
(Lese Datenbank ... 91746 Dateien und Verzeichnisse sind derzeit installiert.)
Vorbereitung zum Entpacken von .../pci.ids_0.0~2021.02.08-1_all.deb ...
Entpacken von pci.ids (0.0~2021.02.08-1) ...
Vormals nicht ausgewähltes Paket libpci3:armhf wird gewählt.
Vorbereitung zum Entpacken von .../libpci3_1%3a3.7.0-5_armhf.deb ...
Entpacken von libpci3:armhf (1:3.7.0-5) ...
Vormals nicht ausgewähltes Paket pciutils wird gewählt.
Vorbereitung zum Entpacken von .../pciutils_1%3a3.7.0-5_armhf.deb ...
Entpacken von pciutils (1:3.7.0-5) ...
Vormals nicht ausgewähltes Paket rpi-eeprom wird gewählt.
Vorbereitung zum Entpacken von .../rpi-eeprom_13.3-1~xbian_armhf.deb ...
Entpacken von rpi-eeprom (13.3-1~xbian) ...
pci.ids (0.0~2021.02.08-1) wird eingerichtet ...
libpci3:armhf (1:3.7.0-5) wird eingerichtet ...
pciutils (1:3.7.0-5) wird eingerichtet ...
rpi-eeprom (13.3-1~xbian) wird eingerichtet ...
Created symlink /etc/systemd/system/multi-user.target.wants/rpi-eeprom-update.service → /lib/systemd/system/rpi-eeprom-update.service.
Trigger für man-db (2.9.4-2) werden verarbeitet ...
Trigger für libc-bin (2.31-xbian2) werden verarbeitet ...
root@kmxbilr4 ~ # 
```

and

```
root@kmxbilr4 ~ # which vcgencmd
/opt/vc/bin/vcgencmd
root@kmxbilr4 ~ # 
```

mkreisl, 2022-01-25:
Problem detected. The reason is that when dpkg is called via apt, a different path is set and therefore the program vcgencmd is not found. Only the APT developers know why this was done.
The path juggling has been bugging me since the beginning of Debian Buster.

closed 2022-01-25 by mkreisl

john-coleman, 2022-01-26:
Thanks for verifying @mkreisl 🙏