xbianonpi/xbian #916: Missing symlink to libbcm_host.so

mensler, 2022-02-03:
I've installed Xbian (release 2022.01.28) on a new RPi4 and installed Hyperion (version 2.0.12 as well as the current nightly).
Starting the hyperiond service fails with the following error message:

`hyperiond: error while loading shared libraries: libbcm_host.so.0: cannot open shared object file: No such file or directory`

I fixed this by creating the necessary symlink manually:

`sudo ln -s /opt/vc/lib/libbcm_host.so /usr/lib/arm-linux-gnueabihf/libbcm_host.so.0`

So I wonder if this is a general issue that should be addressed in future releases.

mkreisl, 2022-02-04:
I noticed that the entire firmware structure of the Raspberry Pi Foundation has been changed (/opt/vc no longer exists for new installations of Pi-OS, but strangely enough it does for my old Pi-OS installation).
This also explains some other strange phenomena. 

This weekend it will be adapted for XBian (xbian-package-firmware) and then everything should be fine again.

mensler, 2022-02-05:
Hi @mkreisl,
sounds good. Do you think this would also solve my new issue?
Now, that Hyperion is working, Kodi does not start with these errors:

failed to open vc4: /usr/lib/dri/vc4_dri.so: cannot open shared object file: No such file or directory
failed to load driver: vc4

Or is this another problem, maybe related to the RPi3 backup that I restored?

mkreisl, 2022-02-05:
When does this error message appear?
It seems to me that this is really total chaos software.

A search for the .so file showed that it can be found here: /usr/lib/arm-linux-gnueabihf/dri/vc4_dri.so
And that is where it belongs

mensler, 2022-02-05:
I got that from /var/log/upstart/xbmc.log:

> Feb 04 22:21:37: MESA-LOADER: failed to open vc4: /usr/lib/dri/vc4_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/arm-linux-gnueabihf/dri:$${ORIGIN}/dri:/usr/lib/dri)
Feb 04 22:21:37: failed to load driver: vc4
Feb 04 22:22:30: MESA-LOADER: failed to open kms_swrast: /usr/lib/dri/kms_swrast_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/arm-linux-gnueabihf/dri:$${ORIGIN}/dri:/usr/lib/dri)
Feb 04 22:22:30: failed to load driver: kms_swrast
start: Job failed to start
Feb 04 22:23:25: MESA-LOADER: failed to open swrast: /usr/lib/dri/swrast_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/arm-linux-gnueabihf/dri:$${ORIGIN}/dri:/usr/lib/dri)
Feb 04 22:23:26: failed to load swrast driver
Feb 04 22:23:26: ERROR: Unable to create GUI. Exiting
xbmc stop/post-start, process 2720
        post-start process 2722

Indeed the lib is there. And for some reason I get no new log entries now when rebooting.

Maybe I should start over with a fresh install - again. When trying to apt-install new packages (bash-completion in this case) I get other weird errors:

> dpkg: nicht behebbarer fataler Fehler, Abbruch:
Laden der Dateilisten-Datei des Paketes »libdrm-nouveau2:armhf«: /var/lib/dpkg/info/libdrm-nouveau2:armhf.list kann nicht geöffnet werden (Eingabe-/Ausgabefehler)
E: Sub-process /usr/bin/dpkg returned an error code (2)

And indeed there are several unreadable files in /var/lib/dpkg/info:

> root@xbian ~ # ls -l /var/lib/dpkg/info/
ls: Zugriff auf '/var/lib/dpkg/info/libdrm-amdgpu1:armhf.list' nicht möglich: Eingabe-/Ausgabefehler
ls: Zugriff auf '/var/lib/dpkg/info/libdrm-nouveau2:armhf.list' nicht möglich: Eingabe-/Ausgabefehler
ls: Zugriff auf '/var/lib/dpkg/info/libdrm-nouveau2:armhf.md5sums' nicht möglich: Eingabe-/Ausgabefehler
ls: Zugriff auf '/var/lib/dpkg/info/libdrm-nouveau2:armhf.shlibs' nicht möglich: Eingabe-/Ausgabefehler
ls: Zugriff auf '/var/lib/dpkg/info/libdrm-nouveau2:armhf.symbols' nicht möglich: Eingabe-/Ausgabefehler
ls: Zugriff auf '/var/lib/dpkg/info/libdrm-nouveau2:armhf.triggers' nicht möglich: Eingabe-/Ausgabefehler
ls: Zugriff auf '/var/lib/dpkg/info/libdrm-radeon1:armhf.md5sums' nicht möglich: Eingabe-/Ausgabefehler
ls: Zugriff auf '/var/lib/dpkg/info/libdrm-radeon1:armhf.shlibs' nicht möglich: Eingabe-/Ausgabefehler
ls: Zugriff auf '/var/lib/dpkg/info/libdrm-radeon1:armhf.symbols' nicht möglich: Eingabe-/Ausgabefehler
ls: Zugriff auf '/var/lib/dpkg/info/libdrm-radeon1:armhf.triggers' nicht möglich: Eingabe-/Ausgabefehler

Something is really broken here, maybe my SD card?

mkreisl, 2022-02-05:
> Something is really broken here, maybe my SD card?

Yes, seems that the FS is completely broken

mensler, 2022-02-10:
Hi,
after reinstallation on a new SD card I can confirm that the file now exists and everything seems to work fine. My xbian-package-firmware version is 10.1-1644084836.

Thanks for your help!

closed 2022-02-10 by mensler