xbianonpi/xbian #918: vcgencmd gone missing?

eremem, 2022-07-07:
Hi there,
did something happened to vcgencmd in xbian? I can recall having been able to use it and after some not so recent updates my installation is missing it. I can't find it in any package via apt-cache either. I'm still using the latest "buster" (/etc/debian_version shows 10.12).

mkreisl, 2022-07-08:
The same problem occurred to me a few days ago. I have no explanation at all. in any case, this program is included in the **xbian-package-firmware** package. 
A reinstall with `sudo apt-get install --reinstall xbian-package-firmware` should solve the problem.

eremem, 2022-07-08:
Indeed, it's quite puzzling. Your solution worked in my case. Thank you for pointing it out.

mkreisl, 2022-07-08:
This is indeed very puzzling. I looked at my installations. The vcgencmd has disappeared from the installations where an Rpi4 is in play. on all older Pi's, the binary is still present.

The-Exnor, 2022-07-08:
> This is indeed very puzzling. I looked at my installations. The vcgencmd has disappeared from the installations where an Rpi4 is in play. on all older Pi's, the binary is still present.

In my case i'm using a Pi3B+ .