xbianonpi/xbian #919: scripts/mod/modpost: Exec format error

Emgalai, 2022-08-26:
I am trying to build this wifi driver on Raspberry Pi B+ (armv6):
https://github.com/brektrou/rtl8821CU

I have latest xbian-package-kernel-headers (5.4.150+-1633004638) installed.

While building the driver, I get an Exex format error about modpost:

xbian@xbian ~/build/rtl8821CU $ make
make ARCH=arm CROSS_COMPILE= -C /lib/modules/5.4.150+/build M=/home/xbian/build/rtl8821CU  modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.150+'
  CC [M]  /home/xbian/build/rtl8821CU/core/rtw_cmd.o
  CC [M]  /home/xbian/build/rtl8821CU/core/rtw_security.o
  CC [M]  /home/xbian/build/rtl8821CU/core/rtw_debug.o
  CC [M]  /home/xbian/build/rtl8821CU/core/rtw_io.o
  CC [M]  /home/xbian/build/rtl8821CU/core/rtw_ioctl_query.o
  CC [M]  /home/xbian/build/rtl8821CU/core/rtw_ioctl_set.o
  CC [M]  /home/xbian/build/rtl8821CU/core/rtw_ieee80211.o
  CC [M]  /home/xbian/build/rtl8821CU/core/rtw_mlme.o
  CC [M]  /home/xbian/build/rtl8821CU/core/rtw_mlme_ext.o
  CC [M]  /home/xbian/build/rtl8821CU/core/rtw_mi.o
  CC [M]  /home/xbian/build/rtl8821CU/core/rtw_wlan_util.o
  CC [M]  /home/xbian/build/rtl8821CU/core/rtw_vht.o
  CC [M]  /home/xbian/build/rtl8821CU/core/rtw_pwrctrl.o
  CC [M]  /home/xbian/build/rtl8821CU/core/rtw_rf.o
  CC [M]  /home/xbian/build/rtl8821CU/core/rtw_recv.o
  CC [M]  /home/xbian/build/rtl8821CU/core/rtw_sta_mgt.o
  CC [M]  /home/xbian/build/rtl8821CU/core/rtw_ap.o
  CC [M]  /home/xbian/build/rtl8821CU/core/mesh/rtw_mesh.o
  CC [M]  /home/xbian/build/rtl8821CU/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /home/xbian/build/rtl8821CU/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /home/xbian/build/rtl8821CU/core/rtw_xmit.o
  CC [M]  /home/xbian/build/rtl8821CU/core/rtw_p2p.o
  CC [M]  /home/xbian/build/rtl8821CU/core/rtw_rson.o
  CC [M]  /home/xbian/build/rtl8821CU/core/rtw_tdls.o
  CC [M]  /home/xbian/build/rtl8821CU/core/rtw_br_ext.o
  CC [M]  /home/xbian/build/rtl8821CU/core/rtw_iol.o
  CC [M]  /home/xbian/build/rtl8821CU/core/rtw_sreset.o
  CC [M]  /home/xbian/build/rtl8821CU/core/rtw_btcoex_wifionly.o
  CC [M]  /home/xbian/build/rtl8821CU/core/rtw_btcoex.o
  CC [M]  /home/xbian/build/rtl8821CU/core/rtw_beamforming.o
  CC [M]  /home/xbian/build/rtl8821CU/core/rtw_odm.o
  CC [M]  /home/xbian/build/rtl8821CU/core/rtw_rm.o
  CC [M]  /home/xbian/build/rtl8821CU/core/rtw_rm_fsm.o
  CC [M]  /home/xbian/build/rtl8821CU/core/efuse/rtw_efuse.o
  CC [M]  /home/xbian/build/rtl8821CU/os_dep/osdep_service.o
  CC [M]  /home/xbian/build/rtl8821CU/os_dep/linux/os_intfs.o
  CC [M]  /home/xbian/build/rtl8821CU/os_dep/linux/usb_intf.o
  CC [M]  /home/xbian/build/rtl8821CU/os_dep/linux/usb_ops_linux.o
  CC [M]  /home/xbian/build/rtl8821CU/os_dep/linux/ioctl_linux.o
  CC [M]  /home/xbian/build/rtl8821CU/os_dep/linux/xmit_linux.o
  CC [M]  /home/xbian/build/rtl8821CU/os_dep/linux/mlme_linux.o
  CC [M]  /home/xbian/build/rtl8821CU/os_dep/linux/recv_linux.o
  CC [M]  /home/xbian/build/rtl8821CU/os_dep/linux/ioctl_cfg80211.o
  CC [M]  /home/xbian/build/rtl8821CU/os_dep/linux/rtw_cfgvendor.o
  CC [M]  /home/xbian/build/rtl8821CU/os_dep/linux/wifi_regd.o
  CC [M]  /home/xbian/build/rtl8821CU/os_dep/linux/rtw_android.o
  CC [M]  /home/xbian/build/rtl8821CU/os_dep/linux/rtw_proc.o
  CC [M]  /home/xbian/build/rtl8821CU/os_dep/linux/rtw_rhashtable.o
  CC [M]  /home/xbian/build/rtl8821CU/os_dep/linux/ioctl_mp.o
  CC [M]  /home/xbian/build/rtl8821CU/hal/hal_intf.o
  CC [M]  /home/xbian/build/rtl8821CU/hal/hal_com.o
  CC [M]  /home/xbian/build/rtl8821CU/hal/hal_com_phycfg.o
  CC [M]  /home/xbian/build/rtl8821CU/hal/hal_phy.o
  CC [M]  /home/xbian/build/rtl8821CU/hal/hal_dm.o
  CC [M]  /home/xbian/build/rtl8821CU/hal/hal_dm_acs.o
  CC [M]  /home/xbian/build/rtl8821CU/hal/hal_btcoex_wifionly.o
  CC [M]  /home/xbian/build/rtl8821CU/hal/hal_btcoex.o
  CC [M]  /home/xbian/build/rtl8821CU/hal/hal_mp.o
  CC [M]  /home/xbian/build/rtl8821CU/hal/hal_mcc.o
  CC [M]  /home/xbian/build/rtl8821CU/hal/hal_hci/hal_usb.o
[... 73 lines cut ...]
  CC [M]  /home/xbian/build/rtl8821CU/hal/phydm/halrf/halrf_debug.o
  CC [M]  /home/xbian/build/rtl8821CU/hal/phydm/halrf/halphyrf_ce.o
  CC [M]  /home/xbian/build/rtl8821CU/hal/phydm/halrf/halrf_powertracking_ce.o
  CC [M]  /home/xbian/build/rtl8821CU/hal/phydm/halrf/halrf_powertracking.o
  CC [M]  /home/xbian/build/rtl8821CU/hal/phydm/halrf/halrf_kfree.o
  CC [M]  /home/xbian/build/rtl8821CU/hal/phydm/rtl8821c/halhwimg8821c_bb.o
  CC [M]  /home/xbian/build/rtl8821CU/hal/phydm/rtl8821c/halhwimg8821c_mac.o
  CC [M]  /home/xbian/build/rtl8821CU/hal/phydm/rtl8821c/halhwimg8821c_rf.o
  CC [M]  /home/xbian/build/rtl8821CU/hal/phydm/rtl8821c/phydm_hal_api8821c.o
  CC [M]  /home/xbian/build/rtl8821CU/hal/phydm/rtl8821c/phydm_regconfig8821c.o
  CC [M]  /home/xbian/build/rtl8821CU/hal/phydm/halrf/rtl8821c/halrf_8821c.o
  CC [M]  /home/xbian/build/rtl8821CU/hal/phydm/halrf/rtl8821c/halrf_iqk_8821c.o
  CC [M]  /home/xbian/build/rtl8821CU/hal/btc/halbtc8821cwifionly.o
  CC [M]  /home/xbian/build/rtl8821CU/hal/btc/halbtc8821c1ant.o
  CC [M]  /home/xbian/build/rtl8821CU/hal/btc/halbtc8821c2ant.o
  CC [M]  /home/xbian/build/rtl8821CU/platform/platform_ops.o
  CC [M]  /home/xbian/build/rtl8821CU/core/rtw_mp.o
  LD [M]  /home/xbian/build/rtl8821CU/8821cu.o
  Building modules, stage 2.
  MODPOST 1 modules
/bin/sh: 1: scripts/mod/modpost: Exec format error
make[2]: *** [scripts/Makefile.modpost:94: __modpost] Error 126
make[1]: *** [Makefile:1650: modules] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.150+'
make: *** [Makefile:2217: modules] Error 2

mkreisl, 2022-08-26:
I don't understand how you can use such an up-to-date wifi stick on a raspberry pi1 fossil.

But anyway. building a module in this way has never worked for me either, but I haven't had much motivation to find out why.

The best thing is to install the 5.15.50 kernel from the **kernel** repository, it's there.
To do this, you have to add **kernel** to **/etc/apt/sources.list.d/xbian.list** as follows:

deb http://apt.xbian.org/ stable main kernel .....

Then **sudo apt-get uodate** and then **sudo apt-get install xbian-package-kernel**

Emgalai, 2022-08-28:
I did that, (and had to also run apt-get upgrade to get Kodi working again).

The module still does not build, the error is similar:

```
...
LD [M] /home/xbian/build/rtl8821CU/8821cu.o
MODPOST /home/xbian/build/rtl8821CU/Module.symvers
/bin/sh: 1: scripts/mod/modpost: Exec format error
make[2]: *** [scripts/Makefile.modpost:134: /home/xbian/build/rtl8821CU/Module.symvers] Error 126
make[1]: *** [Makefile:1786: modules] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.15.50+'
make: *** [Makefile:2217: modules] Error 2
```

> I don't understand how you can use such an up-to-date wifi stick on a raspberry pi1 fossil.

I got this wifi stick cheap and did not intent to use it with Pi, but plans changed. And btw, the PC I'm using now is from 2008, still working fine :smile:

mkreisl, 2022-08-28:
You have not read correctly!

The idea of the kernel upgrade was not to rebuild the module under the new kernel, since a failure was more than predictable, but to directly use the kernel driver 8821au already included in the new kernel package.

```
dpkg-deb -c xbian-package-kernel_5.15.50+-1656696998_armhf.deb | grep 8821
-rw-r--r-- root/root   2720760 2022-07-01 17:39 ./lib/modules/5.15.50+/kernel/drivers/net/wireless/8821au.ko
```

> I got this wifi stick cheap and did not intent to use it with Pi, but plans changed. And btw, the PC I'm using now is from 2008, still working fine smile

This may well be true for PC processors, my self-built NAS is still powered by a Core2Duo from 2007. But the Pi1 is now so lame compared to current Pi's that there is no longer an expression for slowness.

Emgalai, 2022-09-03:
The 8821au driver seems to not be working with the 8821cu. I loaded it with modprobe and re-inserted the wifi stick but still nothing shows up in xbian-config network configuration or iw dev.

mkreisl, 2022-09-04:
Good news :)

Does the driver working now?

I must have succumbed to the Realtek naming confusion, the 8821au driver will hardly run with the 8821cu stick. anyway, yesterday I integrated the 8821cu driver into the standard kernel (version 5.5.61).

Of course I can't test it due to lack of hardware

Emgalai, 2022-09-04:
Yes. It is working fine, at least for now. Thanks for including it in the standard kernel.

closed 2022-10-16 by mkreisl