xbianonpi/xbian #920: xbian-config: "," in SSID breaks wifi network list

Rhino-Cracker, 2022-09-21:
[image]
The 3rd network in the List was called "Ralf-Vodafone-5,0GHz" and WPA2/3 protected (you can use one of both).
xbian-config places everything right from the "," into the column "Type". Don't know what happens to the "5" left from the ",".
Connecting is not possible with this name.

Not necessarily but possibly related to this: My Kodi cannot list wifi networks anymore.

Rhino-Cracker, 2022-09-21:
I've done some pre-work.
You can find the allowed chars for SSIDs here: https://www.cisco.com/assets/sol/sb/WAP321_Emulators/WAP321_Emulator_v1.0.0.3/help/Wireless05.html
According with [that ASCII-Table](https://www.asciitable.com) it results in the following rules:

- 1 first char: %&'()*,-./0-9:<=>@A-Z^_`a-z{|}~
- 0-30 mid-chars: <space> !#%&'()*,-./0-9:;<=>@A-Z^_`a-z{|}~
- 1 end char: !#%&'()*,-./0-9:;<=>@A-Z^_`a-z{|}~
(not sure which of them have to be escabed for bash/grep regex)
(0-9, a-z, A-Z are meant as groups of all digits, letters, while ,-. is meant as the literal chars , - and .)

mkreisl, 2022-10-16:
Two packages are now available in the **staging** repository for testing **(xbian-package-config-shell** and **xbian-package-config-xbmc**).
The problem is that the comma is essential for the internal functioning of xbian-config CLI and cannot be bypassed. So the solution is to UTF-escape the comma in the SSID and convert it back in Kodi's xbian-config GUI. In the xbian-config CLI, however, the comma is displayed as _\u002c_ and this is not changed. It is simply not worth it.

For me, SSIDs that contain a comma are displayed correctly, but I can't connect because my AP (hostapd) simply won't cooperate. The connection does not work with an Android 12 mobile phone either.