xbianonpi/xbian #923: "sudo apt-get update --allow-releaseinfo-change" needed (once) since change in Debian Buster status

superian, 2023-02-16:
Since the change in Buster's status, apt-get update does not get the Buster updates: the apt-secure setup notices that Buster is now 'oldstable' rather than 'stable', and doesn't want to do anything until the user has confirmed that's ok.

The panel you get when logging in as xbian via ssh has a small unscrollable text window, and much of the message saying _why_ it's not working is not visible, so it's necessary to exit that and run `sudo apt update` manually to see why there haven't been any system updates recently.

Running `sudo apt-get update --allow-releaseinfo-change` once makes it work again (at least until Buster changes status again!)

Having that happen should be part of an xbian package update.

mkreisl, 2023-02-16:
One can be of two minds about this. Debian Buster has been replaced by Debian Bullseye almost 2 years ago and is no longer actively supported by XBian, i.e. there are no more updates of binary packages for a little over a year.

> Having that happen should be part of an xbian package update.

Debian distribution upgrade from one version to the next is probably part of the XBian update philosophy. If you don't do it, it's probably the user's problem.

superian, 2023-02-16:
Erm, given that users have been actively warned against using `apt-get dist-upgrade` themselves and I can't spot any suggestion to edit the /etc/apt/sources.list file yourself..

.. what was supposed to have done the upgrade to Bullseye?

Was I supposed to have noticed XBian had changed base distribution - if it has - save XBian settings, download a new ISO and install that?

closed 2023-04-29 by mkreisl