xbianonpi/xbian #925: Install 5.4 Kernel for RPi3b+

sirnacnud, 2023-03-28:
Sorry if this is the wrong place to ask, but I couldn't create a new thread on the forums due to my account being a "newcomer".

Anyways, I have a RPi3b+ which I would like to be able to keep support for HEVC playback which has dropped in Kodi 19 on LibreELEC. I found this [post](https://forum.kodi.tv/showthread.php?tid=369697&pid=3111544#pid3111544) mentioning it works with Kodi 20 with a 5.4 Kernel on Xbian.

How do install a 5.4 Kernel on one of the Xbian releases? I tried the latest RPI4 image and the latest Buster RPI4 image, but both have a newer kernel.

mkreisl, 2023-03-28:
Step 1: Add the kernel area into the file /etc/apt/sources.list.d/xbian.list

```
deb http://apt.xbian.org/      stable  main    kernel    rpi2-...
#deb http://apt.xbian.org/     staging  main    kernel    rpi2-...
#deb http://apt.xbian.org/       devel  main    kernel    rpi2-...
```

Step 2:

sudo apt-get update
sudo apt-get install linux-image-bcm2836=5.4.150+-1633006719

Step 3: 

sudo apt-mark hold linux-image-bcm2836

sirnacnud, 2023-04-01:
Thanks for the instructions, worked perfectly!

closed 2023-04-04 by sirnacnud