xbianonpi/xbian #926: xbian-config does not run with Skin Aeon Nox Silvio

Isayso, 2023-04-23:
If skin Aeon Nox Silvio is activated, xbian-config Addon does not run (exeption missing skin.aeon.nox.silvio folder)

Thank You!

mkreisl, 2023-04-23:
xbian-config GUI addon requires Estuary based skins

This will not be changed either, sorry

closed 2023-04-23 by mkreisl

Isayso, 2023-04-23:
I'm sorry to hear that, I have eye problems and the Estuary skin font is too tiny and the color contrast is weak.

mkreisl, 2023-04-23:
Already tried other colors. I am currently also not sure if you can not also adjust the font size.
Furthermore you only need the xbian-config GUI for setup, it is not needed for normal operation.

Isayso, 2023-04-23:
Yes, they are all white-ish on a light blue, green, red and so on background. Don't see any option beside switching to an Arial font, but no size change. I'm coming from LibreElec, but it seems that they don't fix bugs for RPIs below 4 anymore, so I'm new to xbian. The annoying audio and 60Hz bugs in Libre are gone in xbian, I like that. 

Will use Aeon anyway and if I think I need config, I can switch the skin. Thanks!