xbianonpi/xbian #927: Kodi cannot wake up from screensaver

Isayso, 2023-04-26:
When the screensaver gets activated, Kodi cannot wake up. RPi 3+ with latest xbian. Sometimes also right after start-up.

Error in log:
2023-04-26 18:13:41.665 T:2506    error <general>: Control 9000 in window 10000 has been asked to focus, but it can't

I use Libreelec 11 in parallel and there is no such issue, so the CEC connection is working from the hardware side. But, the remote control is much more responsive in LibreELEC in general..  

Thanks!

Log Error right after start-up the CEC message was on the screen and did not disappear:

2023-04-26 10:16:57.894 T:2567     info <general>: Register - new cec device registered on cec->RPI: CEC Adapter (2708:1001)
2023-04-26 10:16:58.043 T:2506     info <general>: Running the application...
2023-04-26 10:16:58.045 T:2506     info <general>: Loading skin file: Home.xml, load type: KEEP_IN_MEMORY
2023-04-26 10:16:58.260 T:2506     info <general>: starting zeroconf publishing
2023-04-26 10:16:58.265 T:2642     info <general>: ES: Starting UDP Event server on port 9777
2023-04-26 10:16:58.265 T:2642     info <general>: UDP: Listening on port 9777 (ipv6 : false)
2023-04-26 10:16:58.289 T:2506     info <general>: JSONRPC Server: Successfully initialized
2023-04-26 10:16:58.305 T:2506     info <CWebserver[8080]>: Started
2023-04-26 10:17:08.976 T:2573     info <general>: [ script.embuary.helper ] Service: Started
2023-04-26 10:17:08.977 T:2573     info <general>: [ script.embuary.helper ] Service: Player monitor started
2023-04-26 18:13:30.952 T:2506    error <general>: Control 9000 in window 10000 has been asked to focus, but it can't
2023-04-26 18:13:41.665 T:2506     info <general>: Skipped 8 duplicate messages..
2023-04-26 18:13:41.665 T:2506    error <general>: Control 9000 in window 10000 has been asked to focus, but it can't

mkreisl, 2023-04-26:
Which Screensaver are you using?

I have been using a Raspberry Pi 3B+ with XBian as my main media player on a daily basis for about 2 years. It has never happened here that the screen server (Dim) has not woken up.
Sometimes, however, it happens that the CEC connection is disturbed by a Raspberry Pi4 in such a way that Kodi must be restarted and / or the TV must be completely restarted (including disconnection from the power supply). This is probably due to the fact that the devices somehow influence each other negatively.

BTW, Kodi from LE11 is completely different since it doesn't use the outdated raspberry pi interface anymore, which still allows to play HEVC videos on Pi < 4 in XBian.
So the comparison is completely meaningless

Isayso, 2023-04-26:
It is the default screensaver DIM. 

The comparison is not meaningless, the e.g. HDMI cable works from electrical connection side. LE11 (and 10) has horrible audio bugs since Kodi 19, the most usable version for me was 10.0.3, after that the playlists were broken. Even if LE uses Borg technology, I noticed the never fixed bugs. ;)

mkreisl, 2023-04-26:
Does the screensaver not wake up with the default skin? AFAIR you're usually use a different skin
And oh yes, nothing indicates that CEC is not working properly

Isayso, 2023-04-27:
Yes, same thing with Estuary. I can switch on CEC debug messages to see if the remote is still alive.

Isayso, 2023-04-27:
logfile with CEC debug on. Dim takes place at 19:40. Right after I pressed a button the CPU messages in the left corner froze
Shell is alive. 

[kodi - CECafterDIM.zip](https://github.com/xbianonpi/xbian/files/11346525/kodi.-.CECafterDIM.zip)

mkreisl, 2023-04-27:
I only see that the CEC communication is obviously still working. However, I don't see any other messages from Kodi. can this be?

To test, I would disable CEC completely and connect a keyboard or mouse to the Pi.
Just to be sure that it is not the CEC.

But regardless, I don't have the slightest idea what could be the cause of the problem

Isayso, 2023-04-27:
I see Keyboard press events but nothing happens, like CEC.

After ~20s screensaver starts, the cpu and mem overlay in left corner stops to refresh, too.

mkreisl, 2023-04-27:
> After ~20s screensaver starts, the cpu and mem overlay in left corner stops to refresh, too.

So let me get this straight: CEC is guaranteed to be deactivated and several times it works until it freezes

For me this sounds like a hardware problem, otherwise I have no explanation for it

> BTW: You mention that xbian uses some old kernel/driver (?), can this explain why it has no problems with 60Hz videos?

Yes, probably it does.

Isayso, 2023-04-27:
If I switch on debug logging, I get an overlay with mem, cpu, fps in the left hand upper corner, which is changing constantly.
Now, screensaver starts, screen gets dimmed, cpu still changing, I do nothing, suddenly after 20s the cpu update stops

mkreisl, 2023-04-28:
> If I switch on debug logging, I get an overlay with mem, cpu, fps in the left hand upper corner, which is changing constantly. Now, screensaver starts, screen gets dimmed, cpu still changing, I do nothing, suddenly after 20s the cpu update stops.
> 
> System does not wake up at any time point.

That's really scary. I tested it yesterday, and nothing happens. everything is as it should be.

> PS: I think I will do a re-install tomorrow. What is the best method? img file and burn with etcher on SD card?

Image flashing is probably always the best method. since I only have Linux, it's even easier for me with dd :)

> LE seems to be focused on the RPi4, which nobody can buy here.

Yeah. LE is now the (in)official test and development partner for Kodi. And according to the Kodi philosophy no one is interested in what was yesterday. Has always been so

mkreisl, 2023-04-28:
> Error gone! But why? So, I added carefully options and Addons. Still running. Then, I added the gpu_freq value of 500 with overvoltage 2, which was more or less necessary (and recommended) for LE to run at least some 1k@60Hz vids. Boom, no wake-up from dim.
> 
> What a strange effect. The RPi3+ is connected to a power supply from Raspberry, but I also noticed that I see the yellow flash during boot of xbian which I did not see at LE (not sure if they deactivated it).

OMG, I wish you had mentioned earlier that you were overclocking your Pi. That would have saved a lot of time.
Overclocking of Pi3 and Pi3B+ has never worked properly for me, so I leave it alone in the meantime.
Only the Pi4 is well overclockable, provided a good cooling.

> PS: I once tried to send the Kodi team a bug in their player engine, they accused me of using pirate software, which I do not, closed the bug immediately and had some "ignored rules" label for me. Who do they think they are?

I know this all too well. in my experience, the entire Kodi team is a conceited, arrogant bunch.
But of course that's just my (longtime) experience with the people there ;)

Isayso, 2023-04-28:
Funny, I have no problem with overclocking with LE at all.....   I use 3 RPIs in my household, 3 and 3+

mkreisl, 2023-04-28:
Nevertheless, overclocking is always associated with risks!

And I want to tell you that although i am somewhat relieved to know the reason now, I am still totally annoyed about the useless wasted time because of it.

closed 2023-04-28 by mkreisl

Isayso, 2023-04-28:
To know that the upper threshold of the gpu freq, which is only used when needed, can lead to a very specific error, only in Kodi and not for the system (which was still running) is not a waste of time. I would never think of such a side effect, usually the whole system stops, not only one program, or? So, wake up from dim must be very resource demanding.

mkreisl, 2023-04-28:
Then learn something right away:
When you have such a strange error like this, THE FIRST THING YOU HAVE TO DO IS TO DEACTIVATE OVERCLOCKING. something like this should be taken for granted

So, and now the discussion is over, I don't expect any more apologies from you anyway. But I'll remember that, you can believe me!

Isayso, 2023-04-28:
Funny, you complained about rude guys from Kodi and behave in the same way........