xbianonpi/xbian #929: Raspberry PI 3B with the latest Xbian system - read-only USB connected drive.

zalinek, 2023-07-24:
I connected to my raspberry (as in the subject) a hard drive to the USB port. It is visible to other devices in the network environment and I can play multimedia files from it over the network, e.g. via a Windows computer or Android TV with Kodi. But I would also like to be able to write files to this drive over the network, e.g. from a Windows computer. Now when I try to copy some files from my windows computer to this drive, it says that I need permission to perform this operation. How do I give proper xbian permissions to this USB drive so I can copy files over the network from my windows?

zalinek, 2023-07-25:
I know, I found this option earlier and turned it on only after turning it on, I didn't restart the raspberry and it follows that a restart was needed for it to work. After a reboot, writing to the disk works without problems. Thanks for your help and sorry for the confusion.

mkreisl, 2023-07-25:
A reboot would not have been necessary, but is the most convenient solution :smile:

closed 2023-07-25 by mkreisl