xbianonpi/xbian #930: no wifi hardware after upgrade to 6.1.38

StephaneD38, 2023-07-26:
After the upgrade process and the reboot, I loose the wifi connexion to my pi4. The upgrade finished like this:

apt list --upgradable
En train de lister... Fait
linux-image-bcm2836/stable 6.1.38+-1688973008 armhf [pouvant être mis à jour depuis : 6.1.27+-1683653605]
rpi-eeprom/stable 16.0-1~xbian armhf [pouvant être mis à jour depuis : 13.5-1~xbian]
xbian-package-config-shell/stable 20230607-0 armhf [pouvant être mis à jour depuis : 20230503-0]
xbian-package-firmware/stable 11.1-1688293565 armhf [pouvant être mis à jour depuis : 10.1-1644084836]
xbian-package-initramfs-tools/stable 20230704-0 armhf [pouvant être mis à jour depuis : 20220711-0]
xbian-package-xbmc-scripts/stable 2.3.1 armhf [pouvant être mis à jour depuis : 2.2.1]
xbian-package-xbmc/stable 20.2-1688161962 armhf [pouvant être mis à jour depuis : 20.1-1679072453]
xbian-update/stable 20230617-0 armhf [pouvant être mis à jour depuis : 20230419-0]
xbian@Grenoble ~ $ sudo apt upgrade
Lecture des listes de paquets... Fait
Construction de l'arbre des dépendances... Fait
Lecture des informations d'état... Fait
Calcul de la mise à jour... Fait
Les paquets suivants ont été conservés :
  xbian-package-xbmc
Les paquets suivants seront mis à jour :
  linux-image-bcm2836 rpi-eeprom xbian-package-config-shell
  xbian-package-firmware xbian-package-initramfs-tools
  xbian-package-xbmc-scripts xbian-update
7 mis à jour, 0 nouvellement installés, 0 à enlever et 1 non mis à jour.
1 partiellement installés ou enlevés.
Il est nécessaire de prendre 38,3 Mo dans les archives.
Après cette opération, 3 826 ko d'espace disque supplémentaires seront utilisés.
Souhaitez-vous continuer ? [O/n]
Réception de :1 http://apt.xbian.org stable/main armhf xbian-update armhf 20230617-0 [81,8 kB]
Réception de :2 http://apt.xbian.org stable/main armhf xbian-package-xbmc-scripts armhf 2.3.1 [7 118 kB]
Réception de :3 http://apt.xbian.org stable/main armhf xbian-package-config-shell armhf 20230607-0 [965 kB]
Réception de :4 http://apt.xbian.org stable/main armhf xbian-package-firmware armhf 11.1-1688293565 [5 064 kB]
Réception de :5 http://apt.xbian.org stable/rpi2-bullseye armhf linux-image-bcm2836 armhf 6.1.38+-1688973008 [22,7 MB]
Réception de :6 http://apt.xbian.org stable/main armhf xbian-package-initramfs-tools armhf 20230704-0 [26,8 kB]
Réception de :7 http://apt.xbian.org stable/main armhf rpi-eeprom armhf 16.0-1~xbian [2 333 kB]
38,3 Mo réceptionnés en 10s (3 927 ko/s)
Create a snapshot of '/run/xbian-storage/root/@' in '/run/xbian-storage/root/@btrfs-auto-snap_apt-run-2023-07-24-1042'
Delete subvolume (no-commit): '/run/xbian-storage/root/@last_good_known'
Create a snapshot of '/run/xbian-storage/root/@btrfs-auto-snap_apt-run-2023-07-24-1042' in '/run/xbian-storage/root/@last_good_known'
Delete subvolume (no-commit): '/run/xbian-storage/root/@btrfs-auto-snap_apt-run-2023-05-10-1714'
@btrfs-auto-snap_apt-run-2023-07-24-1042, 1 created snapshots, 0 sent snapshots, 1 destroyed snapshots, 0 created filesystems, 0 warnings.
(Lecture de la base de données... 48374 fichiers et répertoires déjà installés.)
Préparation du dépaquetage de .../xbian-update_20230617-0_armhf.deb ...
Dépaquetage de xbian-update (20230617-0) sur (20230419-0) ...
Paramétrage de xbian-update (20230617-0) ...
updating to 1.0 Beta 2
modules volume already exists
updating services
removing non-used services
updating modules
checking interfaces
patching files
patching polkit
Updating to 20230617-0
(Lecture de la base de données... 48374 fichiers et répertoires déjà installés.)
Préparation du dépaquetage de .../xbian-package-xbmc-scripts_2.3.1_armhf.deb ...
Dépaquetage de xbian-package-xbmc-scripts (2.3.1) sur (2.2.1) ...
Paramétrage de xbian-package-xbmc-scripts (2.3.1) ...
(Lecture de la base de données... 48374 fichiers et répertoires déjà installés.)
Préparation du dépaquetage de .../xbian-package-config-shell_20230607-0_armhf.deb ...
Dépaquetage de xbian-package-config-shell (20230607-0) sur (20230503-0) ...
Paramétrage de xbian-package-config-shell (20230607-0) ...
Installing language files ( bg ca cs cs_CZ da el es fr hu hu_HU it ja nl no pl pt pt_BR ru sk sl sv tr zh de en )
(Lecture de la base de données... 48374 fichiers et répertoires déjà installés.)
Préparation du dépaquetage de .../xbian-package-firmware_11.1-1688293565_armhf.deb ...
Dépaquetage de xbian-package-firmware (11.1-1688293565) sur (10.1-1644084836) ...
Paramétrage de xbian-package-firmware (11.1-1688293565) ...
(Lecture de la base de données... 48374 fichiers et répertoires déjà installés.)
Préparation du dépaquetage de .../linux-image-bcm2836_6.1.38+-1688973008_armhf.deb ...
Dépaquetage de linux-image-bcm2836 (6.1.38+-1688973008) sur (6.1.27+-1683653605) ...
Paramétrage de linux-image-bcm2836 (6.1.38+-1688973008) ...
(Lecture de la base de données... 48369 fichiers et répertoires déjà installés.)
Préparation du dépaquetage de .../xbian-package-initramfs-tools_20230704-0_armhf.deb ...
Dépaquetage de xbian-package-initramfs-tools (20230704-0) sur (20220711-0) ...
Remplacement de fichiers dans l'ancien paquet xbian-update (20230617-0) ...
Remplacés par des fichiers du paquet xbian-package-repo (20230425-0) déjà installé...
Paramétrage de xbian-package-initramfs-tools (20230704-0) ...
(Lecture de la base de données... 48369 fichiers et répertoires déjà installés.)
Préparation du dépaquetage de .../rpi-eeprom_16.0-1~xbian_armhf.deb ...
Dépaquetage de rpi-eeprom (16.0-1~xbian) sur (13.5-1~xbian) ...
Paramétrage de rpi-eeprom (16.0-1~xbian) ...
Paramétrage de ipmitool (1.8.18-10.1) ...
usage: update-rc.d [-n] [-f] <basename> remove
       update-rc.d [-n] <basename> disable|enable [S|2|3|4|5]
                -n: not really
                -f: force

The disable|enable API is not stable and might change in the future.
dpkg: erreur de traitement du paquet ipmitool (--configure) :
 installed ipmitool package post-installation script subprocess returned error exit status 1
Traitement des actions différées (« triggers ») pour man-db (2.9.4-2) ...
Traitement des actions différées (« triggers ») pour xbian-package-initramfs-tools (20230704-0) ...
Traitement des actions différées (« triggers ») pour xbian-package-config-shell (20230607-0) ...
Traitement des actions différées (« triggers ») pour xbian-package-repo (20230425-0) ...
Des erreurs ont été rencontrées pendant l'exécution :
 ipmitool
E: Sub-process /usr/bin/dpkg returned an error code (1)
xbian@Grenoble ~ $ uname -a
Linux Grenoble 6.1.27+ #1 SMP PREEMPT Tue May  9 17:01:59 UTC 2023 armv7l GNU/Linux
xbian@Grenoble ~ $ sudo reboot

mkreisl, 2023-07-26:
What does **ipmitool** package?

This does not install properly. Fix this issue first, please

StephaneD38, 2023-08-02:
I have removed ipmitool and re-installed all packages that have been updated and I still have the same issue.
I did also the following command:
```
sudo wpa_supplicant -B -i wlan0 -c /etc/wpa_supplicant/wpa_supplicant.conf
Successfully initialized wpa_supplicant
Could not read interface wlan0 flags: No such device
nl80211: Driver does not support authentication/association or connect commands
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Could not read interface wlan0 flags: No such device
wlan0: Failed to initialize driver interface
```

StephaneD38, 2023-08-02:
lsusb
```
Bus 002 Device 002: ID 152d:0578 JMicron Technology Corp. / JMicron USA Technology Corp. JMS578 SATA 6Gb/s
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 067b:2303 Prolific Technology, Inc. PL2303 Serial Port / Mobile Action MA-8910P
Bus 001 Device 003: ID 7392:7811 Edimax Technology Co., Ltd EW-7811Un 802.11n Wireless Adapter [Realtek RTL8188CUS]
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

StephaneD38, 2023-08-02:
another check during the startup, the new kernel cannot start wifi apparently:

dmesg | grep brcm
```
[    0.758356] irq_brcmstb_l2: registered L2 intc (/soc/interrupt-controller@7ef00100, parent irq: 42)
[    0.762554] brcm-pcie fd500000.pcie: host bridge /scb/pcie@7d500000 ranges:
[    0.762574] brcm-pcie fd500000.pcie:   No bus range found for /scb/pcie@7d500000, using [bus 00-ff]
[    0.762615] brcm-pcie fd500000.pcie:      MEM 0x0600000000..0x063fffffff -> 0x00c0000000
[    0.762654] brcm-pcie fd500000.pcie:   IB MEM 0x0000000000..0x00bfffffff -> 0x0400000000
[    0.763327] brcm-pcie fd500000.pcie: PCI host bridge to bus 0000:00
[    0.815464] brcm-pcie fd500000.pcie: link up, 5.0 GT/s PCIe x1 (SSC)
[    0.906301] brcmstb-i2c fef04500.i2c:  @97500hz registered in polling mode
[    0.906569] brcmstb-i2c fef09500.i2c:  @97500hz registered in polling mode
[    7.024147] brcmfmac: F1 signature read @0x18000000=0x15264345
[    7.073851] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    7.108365] usbcore: registered new interface driver brcmfmac
[    7.115862] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.raspberrypi,4-model-b.bin failed with error -2
[    7.115925] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.bin failed with error -2
[    8.121747] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50
```

mkreisl, 2023-08-02:
> Bus 001 Device 003: ID 7392:7811 Edimax Technology Co., Ltd EW-7811Un 802.11n Wireless Adapter [Realtek RTL8188CUS]

You have a second wifi adapter in the system. what is the purpose of this?

Please show me the output of  `sudo ifconfig` and/or `sudo ip a` (running 6.1.38 kernel)

You are using a   wpa_supplicant.conf  instead of putting the wlan credentials directly into the /etc/network/interface, why ar you duing this

A `dmesg | grep brcmfmac `is not very helpful. If so, please post the complete dmesg after a boot on a pastebin server.

I have observed here that in my various kernel tests (I also frequently switch between 32 bit and 64 bit kernels) the wlan device has suddenly changed (wlan0 became wlan1, for example). This could possibly also be the case for you
if this is the case, you can correct it in the file `/etc/udev/rules.d/70-persistent-net.rules`.

StephaneD38, 2023-08-02:
I have removed the extra wifi adapter, I used it to try to reconnect my pi4 without the need of cable.
this is the dmesg with a working kernel: https://pastebin.com/XVRpQgAu

StephaneD38, 2023-08-02:
apt list --upgradable
```
En train de lister... Fait
linux-image-bcm2836/stable 6.1.38+-1688973008 armhf [pouvant être mis à jour depuis : 6.1.28+-1684360121]
linux-libc-dev/stable 6.1.38-64-1688977949 armhf [pouvant être mis à jour depuis : 6.1.38-1688973008]
xbian-package-initramfs-tools/stable 20230704-0 armhf [pouvant être mis à jour depuis : 20220711-0]
xbian-package-xbmc/stable 20.2-1688161962 armhf [pouvant être mis à jour depuis : 20.1-1679072453]
```
this is the dmesg with 6.1.38 kernel: https://pastebin.com/e4aLQ177
The wifi is not existing again

mkreisl, 2023-08-02:
I think I have found the cause, at least I could reproduce it:

Older kernel packages also contained the wlan firmware for the onboard devices. This is no longer the case, as the **firmware-brcm8021**1 package already contains all the latest firmware.
If a newer kernel package is now installed, the required firmware files are missing.

Solution to the problem:
Reinstalling the **firmware-brcm80211 package** via
`sudo apt-get install --reinstall firmware-brcm80211`

Unfortunately, this cannot be automated and must be done manually, preferably after updating the kernel package before rebooting.

StephaneD38, 2023-08-03:
Good job, it's working now. I have renamed the ticket to help other users, thanks.

SqTER-PL, 2023-08-07:
I still have problem with wifi after kernel update. The last one where it was fine 6.1.28+
WIFI was able to start, but I don't see any network on 5Ghz on channels above 64 (5.320 MHz). The only 5Ghz channels that work well for me after the update are 36 (5.180 Mhz) - 44 (5.220Mhz) - in addition, RPI4 after starting the system connects to the WIFI network for up to two minutes.
I did a clean install of Xbian (kernel 6.1.38+) - didn't help. Any ideas?

mkreisl, 2023-08-07:
@SqTER-PL 

> I did a clean install of Xbian (kernel 6.1.38+) - didn't help. Any ideas?

I don't have 5GHz WIFI here, but luckily my neighbour below me has a Fritzbox with 5ghz WIFI on channel 52 and so I can do some basic tests. 

Here is my result:
1) no matter which kernel (6.1.28 or 6.1.38) is installed here, it always takes the same time until the network is up.
2) with kernel 6.1.28 I see the Fritzbox on channel 52, with kernel 6.1.38 and the firmware from the package firmware-brcm80211 I don't see it.
3) if I use the firmware (and only this one!) from the 6.1.28 kernel package, the Frizbox is visible again with the 6.1.38 kernel. 

This leads to the conclusion that the current official firmware is somehow faulty.

You can now try to get the firmware files from the 6.1.28 package and overwrite them, as you can see below:
```
xbian@xbian /root $ cd /tmp
xbian@xbian /tmp $ apt-get download linux-image-bcm2836=6.1.28+-1684360121
Get:1 http://xbiandeb stable/rpi2-bullseye armhf linux-image-bcm2836 armhf 6.1.28+-1684360121 [23.3 MB]
Fetched 23.3 MB in 4s (6,458 kB/s)              
xbian@xbian /tmp $ dpkg -x linux-image-bcm2836_6.1.28+-1684360121_armhf.deb .
tar: .: Cannot utime: Operation not permitted
tar: .: Cannot change mode to rwxr-xr-x: Operation not permitted
tar: Exiting with failure status due to previous errors
dpkg-deb: error: tar subprocess returned error exit status 2
xbian@xbian /tmp $ sudo cp  lib/firmware/brcm/brcmfmac434* /lib/firmware/brcm/
[sudo] password for xbian: 
xbian@xbian /tmp $ 

```

SqTER-PL, 2023-08-07:
You're right.  Your idea of replacing the firmware works.  
On channels above 64 I don't have any foreign networks, so RPI4 connects much faster.  
Thanks for the bypass.  
It remains to wait for the corrected firmware.

mkreisl, 2023-08-07:
> It remains to wait for the corrected firmware.

It's strange, there has been movement again for the last 2 years https://github.com/RPi-Distro/firmware-nonfree/issues/17