xbianonpi/xbian #931: Can't install Docker

tbertels, 2023-09-09:
On http://forum.xbian.org/thread-4021.html you shared (in 2019) a script to install Docker.
Neither that script nor the one at https://docs.docker.com/engine/install/raspberry-pi-os/ quite work.

To be more precise: they both install Docker, but it can't be run.

When trying to run
`status docker`
we get
`-bash: status: command not found`

And with
`sudo docker run hello-world`
we get
`docker: Cannot connect to the Docker daemon at unix:///var/run/docker.sock. Is the docker daemon running?.`
`See 'docker run --help'.`

In the end of /var/log/docker.log there's
`failed to start daemon: Devices cgroup isn't mounted`

A workaround is to edit /etc/init.d/docker and add
`sudo mkdir /sys/fs/cgroup/devices`
`sudo mount -t cgroup -o devices devices /sys/fs/cgroup/devices`
in the beginning.

`status docker` still doesn't work though.

Is there a way to fix this in a more reliable manner?

mkreisl, 2023-09-09:
```
root@kmxbilr2 ~ # ps ax | grep docker
 7240 ?        Sl     0:00 /usr/bin/dockerd -p /var/run/docker.pid
 7248 ?        Ssl    0:00 containerd --config /var/run/docker/containerd/containerd.toml
 7442 pts/8    S+     0:00 grep --color=auto docker
root@kmxbilr2 ~ # status docker
status: Unknown job: docker
root@kmxbilr2 ~ # service docker status
Docker is running.
root@kmxbilr2 ~ # 
docker run hello-world
Unable to find image 'hello-world:latest' locally
latest: Pulling from library/hello-world
c4018b8bf438: Pull complete 
Digest: sha256:dcba6daec718f547568c562956fa47e1b03673dd010fe6ee58ca806767031d1c
Status: Downloaded newer image for hello-world:latest

Hello from Docker!
This message shows that your installation appears to be working correctly.

To generate this message, Docker took the following steps:
 1. The Docker client contacted the Docker daemon.
 2. The Docker daemon pulled the "hello-world" image from the Docker Hub.
    (arm32v7)
 3. The Docker daemon created a new container from that image which runs the
    executable that produces the output you are currently reading.
 4. The Docker daemon streamed that output to the Docker client, which sent it
    to your terminal.

To try something more ambitious, you can run an Ubuntu container with:
 $ docker run -it ubuntu bash

Share images, automate workflows, and more with a free Docker ID:
 https://hub.docker.com/

For more examples and ideas, visit:
 https://docs.docker.com/get-started/

root@kmxbilr2 ~ # 
```
The reason why **cgroup/devices** is not mounted automatically I can't tell you unfortunately

I don't use docker, I only installed it in a test installation years ago

mkreisl, 2023-09-09:
There is a package called **cgroupfs-mount** that does the directory and mount.
Unfortunately the initscript doesn't work properly either, without rework you can't use that either