xbianonpi/xbian #933: Raspberry PI 3B and the latest Xbian - no image on the video PAL/NTSC output.

zalinek, 2023-11-20:
I have been having problems with Xbian (KODI 20.2) updates on my RPI 3b for some time now. For several days now, attempts to update my Xbian have resulted in error messages. I've had this happen before and then I installed the latest, clean Xbian image on a new SD card using the Xbian installer. As always, I edited the config.txt file by adding the following entry:

sdtv_aspect=1
sdtv_mode=2

in order to get a PAL 4:3 video signal from the video output of my RPI 3B for one of my old TVs. However, this time something is not working, because I do not have a video signal from my raspberry for the old TV - no image on the old TV! I've done this so far and it worked. Are the problems that appeared for me first as errors when trying to update Xbian and then as the lack of PAL video signal for old TVs after a clean installation of Xbian - does this mean the end of support for NTSC/PAL video output in the Xbian system for my Raspberry RPI 3B, for old TVs? Is there any way to fix this problem, or is it only possible to use the HDMI output on the raspberry? My old TV will no longer be supported...
​
Prześlij opinię
Panele boczne

zalinek, 2023-11-20:
So once again, exactly in order. My current Xbian system with KODI 20.2 on my RPI3B generally works well. I have a signal on the composite video output and the image on the old TV is OK. The only problem is that for several days now it has been impossible to install the latest Xbian updates as usual. An update attempt - update all - ends with the following message (content of the /tmp/aptstatus file):

Reading package lists...
Building dependency tree...
Reading state information...
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
  libcec : Depends: python-libcec but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

I once had a similar problem and then, using the Xbian installation tool for windows, I installed the latest, clean Xbian on a separate SD card (from the file XBian_Latest_rpi4.img.gz) and then restored my KODI settings from a previously made backup (home folder) and everything was fine good - composite video output worked and the latest Xbian updates were installed. However, this time - as I wrote earlier - after installing the latest Xbian system image (from the XBian_Latest_rpi4.img.gz file), the composite video output in my RPI3B does not work - the image of the old TV remains dark after running raspberry. And it doesn't matter whether, as I did before, I add the following parameters to the config.txt file:

sdtv_aspect=1
sdtv_mode=2

whether I leave the original config.txt file unchanged according to the instructions. After installing the latest Xbian system image and transferring the SD card with the new, installed image from the reader in the computer to the reader in the raspberry and after starting the raspberry, the video output does not work and the screen of my old TV is black (no image). I tried to present everything step by step again as best as I could. Please help me solve my problem.
​
Prześlij opinię
Panele boczne

mkreisl, 2023-11-21:
I'm sorry, but I still don't know what exactly is going on.
On the one hand you write that XBian runs well, but on the other hand that you don't get a picture on your old tube TV. So which is correct?

You can probably solve the update problem by deleting some CEC packages, they are not needed and only come to the system through the libcec package.
```
sudo apt-get purge python-libcec libcec-dev libcec
```

zalinek, 2023-11-22:
Sorry, I'm writing too much at once and something may be incomprehensible. I will try to write briefly and understandably.

I executed the command in the terminal:

sudo apt-get purge python-libcec libcec-dev libcec

Most of the update packages installed successfully, but two packages still refuse to install and return an error:

cec-utils/stable 6.0.2.1~xbian-1678051568 upgradeable to 6.0.2.1~xbian
libcec6/stable 6.0.2.1~xbian-1678051568 upgradeable to 6.0.2.1~xbian

Trying to update these two packages returns the following error:

Reading package lists...
Building dependency tree...
Reading state information...
The following packages will be DOWNGRADED:
   cec-utils libcec6
0 upgraded, 0 newly installed, 2 downgraded, 0 to remove and 2 not upgraded.
E: Packages were downgraded and -y was used without --allow-downgrades.

Please help me how to fix this update. Thank you in advance for everything.
​
Prześlij opinię
Panele boczne

mkreisl, 2023-11-22:
Everything is already there:
```
sudo apt-get install --allow-downgrades libcec6 cec-utils
```

zalinek, 2023-11-23:
Thank you very much for your help so far - everything is fine with Xbian updates on my currently used card. Everything updated correctly and nothing is stuck anymore.

However, the problem reported in the topic and at the beginning of this thread remains unresolved. On my second SD card I want to install a clean, fresh, up-to-date Xbian image with KODI for my RPI3B and have composite video output work for old TVs. Using the Xbian installer for windows, I install the latest Xbian KODI system image from the image file: XBian_Latest_rpi4.img.gz on a formatted SD card.
After the image installation process is completed, I move the SD card from the reader in the computer to the reader in my raspberry and run it with my old TV connected using the composite video output in my raspberry. However, the system does not start - no image on my old TV. The composite video output on my raspberry is not working - no signal? A few months ago, I installed Xbian on my RBI3B this way and the composite video output worked normally with my old TV. Please help me on this topic.
​

mkreisl, 2023-11-23:
Please don't be angry with me, but I have neither the time nor the inclination for such a troubleshooting, If it is one.
you have a working sd card, make a copy of it and that's it