xbianonpi/xbian #935: Addon xbian xbmc-upstart bridge incompatible after latest update

JoonasC, 2023-12-12:
Kodi updated automatically, and now I get a message telling me that xbian xbmc-upstart bridge is incompatible. The Kodi version is 20.2. Is this safe to ignore?

JoonasC, 2023-12-12:
Actually, it seems like it was the addon that was updated, not Kodi:

Start-Date: 2023-12-11  23:49:29
Commandline: /bin/unattended-upgrade -d
Upgrade: xbian-package-upstart-xbmc-bridge:armhf (2.0.5, 2.0.5)
End-Date: 2023-12-11  23:49:31

mkreisl, 2023-12-12:
I had that too, but I hadn't figured out exactly what the problem was. I guess it was because there were two versions of different architectures (armhf and all) in the repository. I then deleted the armhf version in the repo and refreshed the package with

```
sudo apt-get update && sudo apt-get install --reinstall xbian-package-upstart-xbmc-bridge 
```

again. The problem then disappeared

JoonasC, 2023-12-12:
I tried refreshing the package but it didn't help. What do you mean by deleting a version in a repo?

mkreisl, 2023-12-15:
Hmmm, I'm not sure. as I said, I didn't inspect the problem any further because it disappeared again immediately, especially as I was working on something else (crosscrading 32bit->64bit) and therefore there were a lot of changes at once.
 
And the problem didn't occur with other installations either. So it seems to be something special for you. a look at the Kodi log should actually help.

> What do you mean by deleting a version in a repo?

In the course of the 64bit story the whole repository had to be restructured as almost all packages were hardcoded to armhf, even packages that are definitely not architecture dependent (scripts, python and the like). this led to duplicate packages (armhf and all) even though the content is identical.

JoonasC, 2023-12-17:
The Kodi log contains the following, which doesn't really tell me much else:
2023-12-18 00:30:05.239 T:2221     info <general>: CAddonMgr::FindAddons: script.service.xbian.upstart-bridge v2.0.5 installed
2023-12-18 00:30:09.238 T:2766     info <general>: ADDON: script.service.xbian.upstart-bridge version 2.0.5 is incompatible
2023-12-18 00:30:09.239 T:2766  warning <general>: ADDON: failed to disable script.service.xbian.upstart-bridge

JoonasC, 2023-12-17:
Although it looks like my attempt at refreshing the package ended up reinstalling the armhf version:
Reinstall: xbian-package-upstart-xbmc-bridge:armhf (2.0.5)

How can I tell apt to remove the armhf version and replace it with all?

mkreisl, 2023-12-18:
Something strange is happening. I just looked at the main repository and the two mirrors and there is only the **all** version of the 2.0.5 version of the package. So what's going on with you?
The only thing I can think of is that the package is still cached in your local directory

```
sudo apt-get clean
sudo apt-get update
```
before you reinstall the package

JoonasC, 2023-12-18:
Doing that didn't help. Although I noticed something strange.

The /var/log/apt/term.log file reads:
Log started: 2023-12-18  23:12:26
(Reading database ... 41820 files and directories currently installe
d.)
Preparing to unpack .../xbian-package-upstart-xbmc-bridge_2.0.5_all.
deb ...
Unpacking xbian-package-upstart-xbmc-bridge (2.0.5) over (2.0.5) ...
Setting up xbian-package-upstart-xbmc-bridge (2.0.5) ...
Log ended: 2023-12-18  23:12:27

While the /var/log/apt/history.log file reads:
Start-Date: 2023-12-18  23:12:26
Commandline: apt-get install --reinstall xbian-package-upstart-xbmc-
bridge
Requested-By: xbian (1000)
Reinstall: xbian-package-upstart-xbmc-bridge:armhf (2.0.5)
End-Date: 2023-12-18  23:12:27

JoonasC, 2024-01-02:
Do you have an idea what could be causing this?

The-Exnor, 2024-01-03:
I now have a similar situation.  
After updating (apt update, upgrade, etc...) , every time Kodi starts it gives me a message that the xbmc-upstart bridge is incompatible.  
I just press ok, and everything seem to work so far.

Some info about the version of the OS:

PRETTY_NAME="XBian 11.0 - Bullseye"
NAME=XBian
VERSION_ID=11.0
VERSION="XBian 11.0 - Bullseye"
ID=raspbian
ID_LIKE=debian

JoonasC, 2024-01-16:
The latest update spontaneously fixed this for me

mkreisl, 2024-01-16:
That's really strange. I have no explanation for either the appearance or the disappearance of the problem