xbianonpi/xbian #936: Overwriting of cmdline.txt on updates

AverageUser000000, 2023-12-18:
Every update where the cmdline.txt file is touched wipes out any previous version.

I use the RPI4 with Xbian and because of the SSD attached (well it's controller, actually) UAS causes it to die eventually so there's a quirk on the cmdline.txt to fix it:
usb-storage.quirks=XXXX:XXXX:u

Is there a way of preserving this when you do the update? A merge of the cmdline.txt files from the existing system and your new one?

mkreisl, 2023-12-18:
Yes, this is obviously a small problem that affects one or the other user
The **xbian-update** package is responsible for this, I had already added a few parameters some time ago which are kept. Unfortunately this is hardcoded so far and cannot be changed by the user.
I have now extended this :smile:
With the new **xbian-update** package which will be available soon the list becomes parameterizable via the file **/etc/default/xbian-internals**, parameter OPT_CMDLINE_ARGS
You can add the following text to this file, so that everything works with the next update of the xbian-update package
```
# Here you can specify parameters that should be kept in the
# /boot/cmdline.txt file. Please note that parameters that require
# an argument must end with an =.
#
OPT_CMDLINE_ARGS='partswap vnc cma= ip= cnet= modules-load= usb-storage.quirks='
```
Your missing parameter is already entered there

AverageUser000000, 2023-12-18:
Wow, great news!
I did not know of the existence of this file and its options. Is this documented somewhere? 

I have pre-prepared my /etc/default/xbian-internals with the parameter.

This will make a great change from the constant problem of me forgetting to edit the cmdline.txt file after an update and before reboot and then finding the SSD dies... Eek. 

I am surprised I am the first to mention this!
I've been manually fixing it for a while now so perhaps many are doing the same? Either that or I am the only one running RPI4, Xbian and SSD on a non-UAS USB-to-SATA controller  (even though there's an entire huge thread about it on the RPI forums).

Thanks again, Man!

mkreisl, 2023-12-19:
> I did not know of the existence of this file and its options. Is this documented somewhere

Documenting is just one of those things. there's no one there to do it
By the way, this is for internal purposes, for emergencies so to speak