xbianonpi/xbian #937: Repeated lockups - related to having two memory sticks??

superian, 2024-03-10:
Running on a Pi B 2. If I ssh in, the version bits of the header say this:

Linux xbian 6.1.66+ #1 SMP PREEMPT Sat Jan  6 23:21:01 UTC 2024 armv7l
XBian 12.0 - Bullseye - - Bleeding Edge, 2012-2024

Doing an update/upgrade via the menu system tells me I am up to date with packages, but the most recent update was a week ago.

Since then, the system has been locking up - logged in users get messages like "client_loop: send disconnect: Broken pipe" before the session is ended, the TV the Pi is attached to goes blank, the Kore remote control app disconnects, and attempts to connect over ssh again are refused.

It's happened frequently: rsync a file to a memory stick attached to the Pi and it will work resulting in a complete file at the Pi end.. but then the Pi locks up.

I've just watched a couple of things on it via Kodi. The first one went fine. When I stopped the second video, it locked up.

If I take the two memory sticks out, it seems to be OK in terms of logging in and out and in again, but as I have the media on memory sticks, it's not particularly useful.

Because of the way it locks up, the log files are truncated and so not helpful.

Any suggestions?

mkreisl, 2024-03-11:
Sounds like a power problem. Use a different power supply unit

superian, 2024-03-11:
It's looking like a microSD card issue: the lockups got worse (happened in the middle of watching something, twice).

Attempting to overwrite the card with a fresh image failed.

Using that image on a new card looks to be OK so far...

mkreisl, 2024-03-12:
Of course that could be the case. I had a similar problem a few weeks ago, where the file system was breaking down more and more. I suspected the sd-card at the time, but that was not the case. It was a dying power supply, which also affected the file system. after the power supply was replaced and the backup was flashed to the sd-card again with a backup, the problem was solved for me :smile:

superian, 2024-03-12:
Doing some more tests shows the old microSD card is thoroughly stuffed - can't write to it at all.

closed 2024-03-12 by superian