xbianonpi/xbian #938: Screen gives no signal after Kodi startup

hewitje, 2024-03-18:
I had to reinstall my Xbian due to a SD card issue. But now I have a new problem. 
When I fresh install all seems normal, I start up, I get the wizard, and I can use Kodi. But after reboot, I get the kodi splash screen, and a fraction of a second I see the Kodi version screen you see right before kodi starts up. But after that my screen gives no signal. I tried different screens, different cables. I notice the device is still on, I can access files over network, and when I press the combination of buttons I know would reboot the device on the remote this would also still do. So it seems kodi is running, but I just can not see it. I tried reinstalling, but same problem. After I tried reinstalling a slightly older version (1st of march), but still same problem.

I use a raspberry pi 4B with 8gb of Ram.

mkreisl, 2024-03-18:
The problem sounds familiar to me, it is important that you boot with an active initramfs. If this is not the case, this problem can occur.

Which image did you use?

mkreisl, 2024-03-18:
`initramfs initramfs.gz followkernel`

this line must not be commented out in config.txt
This file is located on the first partition of the sd-card

hewitje, 2024-03-18:
This solved the issue indeed, but only for the first boot.
After rebooting again, the problem reoccurs again. Is there a reason this line is commented out every time with booting?

mkreisl, 2024-03-18:
No, of course not. So far it was only known that this occurs with Kodi v21 for unknown reasons. This has been fixed with this update https://github.com/xbianonpi/xbian-package-initramfs/commit/03e336cb945facee864102c28b50e0ab3329ed77
I was not previously aware that it also occurs with Kodi v20.

You can fix this by forcing the use of initramfs in the XBian-config GUI within Kodi or setting manually in file **/etc/default/xbian-initramfs** the variable **FORCEINITRAM=yes**

By the way, this only happens with a Raspberry 4, but not with a Raspberry Pi5.
This is all quite strange

hewitje, 2024-03-18:
Thanks, this solved the issue I had.

closed 2024-03-18 by hewitje