xbianonpi/xbian #941: Logrotation.

Steven-Jones911, 2025-02-04:
I am not sure if this is the right place to offer up this problem.

When log files are rotated, they keep being written to the old log, that has been renamed.

So, for example, **messages.log** is renamed to **messages.log.1** and a new **messages.log** is created. The logger still writes to the old **messages.log.1**

In /etc/logrotate.d/rsyslog is the option:
```
postrotate
            /usr/lib/rsyslog/rsyslog-rotate

```
In rsyslog-rotate is the command:
`            invoke-rc.d rsyslog rotate > /dev/null
`

When run, it barfs at "rotate".

Having changed it to "reload", this fails to work also, so I changed it do "restart" and it successfully rolls the logs over and begins using the new log.

Looking at the init script for rsyslog, _rsyslog.conf_, it has no option for rotate.

In another system I have, Devuan on RPI3, it has (amongst other options):
 ```
rotate)
        log_daemon_msg "Closing open files" "$RSYSLOGD"
        do_rotate
        log_end_msg $?
        ;;
```

mkreisl, 2025-02-07:
The problem is that since Debian Bookworm there is no more Sys-V initscript in the rsyslog package

Steven-Jones911, 2025-02-07:
Ok, I thought so as well but it didn't when I tried it, it continued to write to the old log. Perhaps something else is going on.

mkreisl, 2025-02-08:
Could you check if you have that file:

`/etc/init/rsyslog.conf`

Steven-Jones911, 2025-02-08:
Yes,
`description "enhanced syslogd"
author      "mkreisl mkreisl@xbian.org"

start on started xbian-run
stop on starting rc RUNLEVEL=[06]

#respawn

env DAEMON=/usr/sbin/rsyslogd

pre-start script
    test -e $DAEMON || { stop; exit 0; }
end script

script
    $DAEMON -n -iNONE
end script

`

Steven-Jones911, 2025-02-08:
I just checked the logs and the logs have been rotated, this morning, without any modification to any scripts by me.

Since I reported this issue, the Pi4 has been rebooted so this might have fixed whatever was ailing it.

Weird.

mkreisl, 2025-02-08:
Strange, maybe cron jobs were locked

Steven-Jones911, 2025-02-09:
Who knows?
I will close it as solved. Just another mystery of the universe... 😊

closed 2025-02-09 by Steven-Jones911