xbianonpi/xbian #945: Using Gigabit Networking on Raspberry Pi 3 with Xbian and KODI

evest10, 2025-03-16:
I am currently using a Raspberry PI 3B with the updated Xbian system with Kodi installed. I have a question: if I buy a similar USB 3.0 to gigabit ethernet adapter for my RPI 3B, for example from this link:

https://allegro.pl/oferta/przejsciowka-usb-ethernet-adapter-karta-sieciowa-lan-gigabit-rj45-1000-mbps-15522682024

will I get speeds above 100 Mb/s? Supposedly, this is possible according to this article:

https://www.jeffgeerling.com/blogs/jeff-geerling/getting-gigabit-networking

So if I connect a similar adapter to my RPI 3B via USB, will I immediately get higher speeds than the current 93-94 Mb/s or do I have to configure something else - add something? According to the article from the above link, the interface should be properly configured by editing and adding a line for the new eth adapter. If this is necessary, please provide detailed step-by-step instructions on how to do this on my Raspberry PI 3B with the Xbian system with Kodi using a console operated with "putty". If anyone could help me, I would be very grateful.

evest10, 2025-03-18:
Thank you for the information and I am already thinking about purchasing more powerful equipment.

closed 2025-03-18 by evest10

mkreisl, 2025-03-19:
> Thank you for the information and I am already thinking about purchasing more powerful equipment.

That is a very good decision. I didn't suggest this to you because I don't know your financial situation.