xbianonpi/xbian #948: Error starting XBian-Config after installation from the current image (2025.04.04)

eremem, 2025-04-11:
Starting Xbian-Config shows an error pop-up.

~/.kodi/temp/kodi.log:
```
2025-04-11 12:04:52.939 T:5811    error <general>: EXCEPTION Thrown (PythonToCppException) : -->Python callback/script returned the following error<--
 - NOTE: IGNORING THIS CAN LEAD TO MEMORY LEAKS!
Error Type: <class 'SyntaxError'>
Error Contents: from __future__ imports must occur at the beginning of the file (default.py, line 2)
  File "/usr/local/share/kodi/addons/plugin.xbianconfig/default.py", line 2
    from __future__ import print_function
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
SyntaxError: from __future__ imports must occur at the beginning of the file
-->End of Python script error report<--

```

mkreisl, 2025-04-11:
Some information about your installation would be extremely helpful, Otherwise this won't work here

eremem, 2025-04-11:
Well, current rpi5 64 bit image downloaded today, it's just a fresh installation, nothing more...
There seems to be a newer version of the package xbian-package-config-xbmc provided in that image.
I downgraded the package to the version working on my older installation (20231115) and this one works also with my new installation from the current image.

mkreisl, 2025-04-11:
You are probably using the 64 bit image. The devel repo has been activated by mistake and therefore the package xbian-package-config-xbmc has been installed from this devel repo. However, it looks like this only works with Python 13 from Debian Trixie.
Changes were necessary because some Python packages are no longer available in Debian Trixie

mkreisl, 2025-04-11:
Ah, you've already done it :smile:
The new image that is being built on today has been fixed in the meantime

mkreisl, 2025-04-11:
But what irritates me: on my 64bit Bookworm installation this new devel package is also installed and there the package works, but with Kodi 22 Alpha from the beginning of April

All very strange

eremem, 2025-04-11:
Maybe there is some difference to the python interpreter the next kodi version uses?

mkreisl, 2025-04-11:
Hard to imagine. But I will install a current Kodi 21 again this weekend for comparison. This has to be done anyway because the new current kernel 6.12 requires an ffmpeg patch so that the hardware decoding works again